InfiniTimeOrg/InfiniTime #703: [Feature Request] Reorder/hide apps

adocampo, 2021-09-29:
I think all apps are great, and I thank all devs for creating such a great piece of software.

But being realistic, for me there are apps more interesting than others, so I don't want to scroll down (now two pages) to configure an alarm while navigation, which I don't even know how to use it is on the first page. 

This will become even more frustrating as more apps will be on the watch, so I think a way to reorder or hide/show apps will be a nice to have feature.

calhix, 2021-09-29:
Having the ability to hide apps and move them around would be huge, out of the 12 I only find 6 useful in my case and they are all over the place.

dyamon, 2021-09-30:
I think one of the big todos atm is the ability to dinamically load apps into the watch without the need to recompile the firmware. The decoupling of firmware and apps will most likely solve the request of hiding apps, so I don't think it is wise to implement some patch code to simulate this now.

The reordering feature on the other hand is nice to have, and we would be able to reuse this code later on as well. That said, afaiu the app menu is really two different pages of apps. To simplify the task we would probably need to redesign the menu code.

In the meantime, if you are able to compile the firmware yourself (can help with that as well), a temporary solution would be to modify the code here:

https://github.com/JF002/InfiniTime/blob/e9bb0b3cdd0e1b95b7bfdcfb94db5a6a0966eed7/src/displayapp/screens/ApplicationList.cpp#L40-L66

The two lists of 6 elements determine the order of the apps, so moving a line around will result in a different order.

Changing one of the elements in the array to

```
{"", Apps::None},
```

will "disable" the app, resulting in an empty spot in the menu.

adocampo, 2021-09-30:
> The reordering feature on the other hand is nice to have, and we would be able to reuse this code later on as well. That said, afaiu the app menu is really two different pages of apps. To simplify the task we would probably need to redesign the menu code

What pebble did was let the phone companion app reorder (well, and install and uninstall) the app list, so it synced the desired order back to the watch. I think that's quite smart as it let the footprint of the firmware quite small,  if there is an "official" InfiniTime companion app, I doubt Gadgetbridgo or Siglo can have this in their roadmap. 
Without a proper companion app, it should be done via the watch itself, and as I don't know the internals at all, I can just suggest a semi-complex settings page where all the apps are listed and a number on the side indicating the position. (I don't even know if there is a drag and drop capability or if it would even be practical due to the screen size)

I need to find some time to play with fw compiling, between my work, side projects and my baby, I have barely time to rummage with anything, but I would like to try it, I'll keep you informed.

Avamander, 2021-09-30:
> I doubt Gadgetbridgo or Siglo can have this in their roadmap.

GB has it for the Pebbles (and Bips I think) already, so it's a matter of wiring things up, as usual. There's nothing that stops Siglo or Amazfish implementing a similar feature.

adocampo, 2021-09-30:
Oh, didn't know GB had this feature! I used the official Pebble app as I find it simpler and nicer than GB. That's great then, as you say, that's only matter of wiring things up to let companion apps access to menu configuration.

JF002, 2021-09-30:
As we are adding more and more settings to the project, I also feel like we won't be able to fit all settings into the UI of InfiniTime. Using the companion apps for more complex settings looks like a good idea. We would indeed need to add the functionality in InfiniTime (dynamic ordering of the app) and a configuration BLE service. Then, we'll need support from the companion apps. I'm not too worried about this one, as we are good friend with teams from Gadgetbridge, Amazfish, Siglo and Infini-iOS :)

kieranc, 2021-10-01:
I agree with the sentiment that reordering the apps would me a better approach than, for example #708 sadly, as adding a new customisable app list seems unnecessary when we could just make the existing one customisable... Maybe in the mean time we could reorder the list, Navigation and step count probably don't need to be page 1, that way we can more the alarm app up.... and maybe 2048? :-)

ncartron, 2021-11-12:
I agree with @kieranc - for the time being, moving the less used apps to the 2nd page would do :)