InfiniTimeOrg/InfiniTime #707: Bluetooth connection still dropping out

Bugsbane, 2021-09-30:
So, I updated to version 1.6, but I still have the issue where after a number of hours that the bluetooth connection drops out, and Gadgetbridge can't reconnect until the watch is restarted. After restarting, pairing works fine.

Steps to reproduce:
1. Pair watch to phone
2. Wait 12-24 hours.
3. Watch bluetooth connection resets.

Expected behavior:
Connection remains active.

Specs:
Infinitime 1.6 with 1.0 bootloader flashed by Gadgetbridge 0.59.3 on Android 10 on a Samsung Galaxy A51. Is there any other info that would be useful?

omartijn, 2021-10-01:
I am having a similar issue. For me, it drops connection _all_ the time, but _that_ at least seems to be due to Sailfish power-management, since it doesn't happen when the screen remains on (and no, setting the suspend policy didn't help).

However, after a day of constant connection dropping and reconnects, the watch stopped accepting connections. Every time I tried to connect from the phone side, it would briefly show connection and then give an error that the passcode was incorrect. The only way to fix it was rebooting the watch.

kieranc, 2021-10-01:
It is looking like the heart rate monitor is causing Bluetooth issues, are either of you using the HRM, or not?

hallebubbsan, 2021-10-02:
> It is looking like the heart rate monitor is causing Bluetooth issues, are either of you using the HRM, or not?

I started the hrm at the same time I got the connection issue

omartijn, 2021-10-02:
I didn't notice the link at the time, but I do remember at some point in the evening turning it on for a little bit and then switching it off. Later I did experience the connectivity issues. I can't be sure of the exact time, but it sounds plausible that it's the culprit.

akutuzov, 2021-10-02:
> It is looking like the heart rate monitor is causing Bluetooth issues, are either of you using the HRM, or not?

I notice exactly the same. Without HRM, the watch can stay connected for whole days (with occasional re-connects). As soon as HRM is turned on, the Bluetooth connection dies in a matter of an hour if not less. Only resetting the watch helps then.

InfiniTime 1.6.0, Sailfish OS 4.2 on Sony Xperia 10

RichardNewb1e, 2021-10-03:
I'm having same issue on Android 10 of XiaoMi 9 without using HRM. The first 12 hours or so worked fine, then Bluetooth was dead and no longer discoverable.

Avamander, 2021-10-03:
@RichardNewb1e 

> I'm having same issue on Android 10 of XiaoMi 9 without using HRM. The first 12 hours or so worked fine, then Bluetooth was dead and no longer discoverable.

Up-to-date InfiniTime?

RichardNewb1e, 2021-10-03:
> @RichardNewb1e
> 
> > I'm having same issue on Android 10 of XiaoMi 9 without using HRM. The first 12 hours or so worked fine, then Bluetooth was dead and no longer discoverable.
> 
> Up-to-date InfiniTime?

My bad, version was 1.4. I did update it but turned out that I forgot to validate it or I probably pressed reset :( Should I delete my previous comment? It looks silly.

Avamander, 2021-10-03:
@RichardNewb1e 

> My bad, version was 1.4. I did update it but turned out that I forgot to validate it or I probably pressed reset :( Should I delete my previous comment? It looks silly.

No, keep it, someone else might make the same discovery.

bakeromso, 2021-10-05:
I've had the same issue happen twice now (connection working fine for over a day and suddenly watch is undiscoverable until watch reset)
I think I might have used HRM on both occasions, but I'm not sure. I'll stay away from using HRM the coming days and will report if this prevents the connection issue

arjanvlek, 2021-10-06:
Same issue here. Thought it was caused by the infini-IOS 0.9.1 update, but nRFConnect also failed to connect to the watch.

For me it also happend after having used the HRM. Before using the HRM, the watch could connect to bluetooth for multiple days without needing to be reset.

adocampo, 2021-12-15:
This still persists in 1.7.0. When I pair the watch with Gadgetbridge, and leave the bluetooth reach, there is no reconnection back when I near, no matter the time I wait for it. I _always_ need to delete the watch and re-pair it again, making pairing almost useless.

Avamander, 2021-12-15:
@adocampo Until bond persistence is released the advice has been to not pair, you're given the choice when adding the device to GB.

tootbrute, 2022-01-11:
> @adocampo Until bond persistence is released the advice has been to not pair, you're given the choice when adding the device to GB.

Thank you! I'm a first week user of Pinetime and this helps so much!

adocampo, 2022-01-12:
I hate to be that guy, but this is still present on 1.8.0, which if I understood correctly what it says on the announcement, shouldn't be happening now.
Rebooting the watch does nothing, just re-pairing with GB seems to connect back again.
I'm on Infinitime 1.8.0, Android 10 and GB 0.64

kieranc, 2022-01-12:
> I hate to be that guy, but this is still present on 1.8.0, which if I understood correctly what it says on the announcement, shouldn't be happening now. Rebooting the watch does nothing, just re-pairing with GB seems to connect back again. I'm on Infinitime 1.8.0, Android 10 and GB 0.64

So you're getting into a situation where the phone can't see the watch, rebooting the watch does nothing but removing the bond in gadgetbridge and recreating it allows you to reconnect.
A few possibilities:
1) airplane mode on phone - if you toggle bluetooth or airplane mode, your phone MAC changes and causes problems - i'm not sure if this is resolved in 1.8, if it is, only when using secure pairing with PIN.
2) HRM - if you have the heart rate sensor active it seems to interfere with the bluetooth connection, part of another ongoing bug
3) multiple connections - if the watch is paired in android and GB it can cause problems, this can occur automatically when doing secure pairing I think?
Do any of these apply?

adocampo, 2022-01-13:
Well, after a while using it, I'm not quite in the same situation as described, and I must admit BLE reconnection works way better on this version, but not perfect, at least in my scenario, but I'm realizing it is because of GB. Let me explain it.

 - When I'm out of the reach of the BLE of the phone, the watch disconnects. If I come back, it connects back just as expected.
 - If I'm connected, and reboot the watch, it reconnects just as expected on the watch start.
 - If I'm connected, and reboot the phone, it doesn't reconnects just as expected on the phone start because GB doesn't start automatically, I need to manually open the application (I have "Start automatically" checkbox on GB settings, but it doesn't seem to work)
 - If for some reason, GB is closed (ie: the icon in the notification bar is not present), connection is lost.

So, I would say BLE on InfiniTime works as it should, but for some reason, GB is quite unstable on my phone and seems to close by itself now and then, killing my watch functionality...

kieranc, 2022-01-13:
Is it possible that your phone has some sort of battery saver which is killing background tasks?

adocampo, 2022-01-13:
The only thing my phone has is it shuts down at deep night (around 2am) and turns on around 6am, the rest of the day it has no restrictions at all. I don't really know why it does that, I don't remember to mess with energy savings at all, but I will take a look into that.

kieranc, 2022-01-15:
Are you using secure pairing (PIN) or not? If you're not, rebooting the phone will cause the phone MAC to change and reconnect won't work, you'll have to reconnect manually in GB.

I just tested toggling bluetooth with secure pairing and it does reconnect automatically, I had to enable the option in GB to reconnect on BT activation but otherwise it worked as expected, and I think it would also work when rebooting the phone. To enable secure pairing you need to choose 'pair' rather than 'don't pair' when adding the watch in GB, it should ask you for a PIN which will be displayed on the watch.

adocampo, 2022-01-15:
Yes, I paired with PIN and it reconnects just nice IF GB app is open, which is not always the case. But as I said, it seems the culprit for closing GB must be my phone

Avamander, 2022-01-15:
@adocampo CDM pairing in settings can make it easier for GB to restart when killed by your phone. However if your phone is killing a notification listener accessibility service, then that sounds like OOM and can't trivially be avoided.

kieranc, 2022-01-16:
> Yes, I paired with PIN and it reconnects just nice IF GB app is open, which is not always the case. But as I said, it seems the culprit for closing GB must be my phone

Is your phone running stock android or lineage/aosp etc? Which manufacturer? It sounds to me like GB is being killed in the background, there might be a setting to prevent it.

adocampo, 2022-01-16:
It's a POCO X3 Pro with Xiaomi's custom ROM MIUI 12. It has 6GBr RAM and I don't use so many apps: just mail, maps, browser, plex media player and PlexAmp, Youtube, Home Assistant, Telegram, and RocketChat. I have others like Termux, Netlfix, HBO, Prime Video, Disney+ but I use them very sporadically. I have even more than I barely use anytime. So I wouln'd bet for a OOM Kill.

I noticed GB doesn't starts automatically, so every night's halt stops GB and it never starts again. I'm going to disable that energy saving option to see if there is something wrong with it and GB.

Avamander, 2022-01-16:
MIUI has one of the most aggressive BG service killers, do check what you have in your ultra power saving mode settings and in app's settings under battery optimizations. MIUI also has a separate permission for autostart you can toggle, so check that as well.

the-voidl, 2022-01-20:
I am on Mi9T with LineageOS.
Every night I enable flight mode on my phone so the watch (PineTime) disconnects.

Every morning I cannot reconnect the watch. The only way is to completele delete it from Gadgetbridge and re-pair it.

Avamander, 2022-01-20:
@the-voidl What InfiniTime version, are you pressing pair when asked?

the-voidl, 2022-01-20:
> @the-voidl What InfiniTime version, are you pressing pair when asked?

I originally paired the watch with flashed IT v1.6.0

On the advice of a friend, I paired the device again today under version 1.8.0 without removing it from Gadgetbridge first.
I will keep you updated if that already solved the problem.

JF002, 2022-01-20:
> I originally paired the watch with flashed IT v1.6.0

InfiniTime did not support bonding in 1.6.0. So it might help to remove your watch from Gadgetbridge and re-pair it (accept pairing when asked) so your watch and your phone can bond correctly (using the PIN code).

the-voidl, 2022-01-30:
I can confirm that this is fully resolved in 1.8.0 :)

JF002, 2022-01-31:
I guess we can close this issue. Feel free to repoen or open a new one in case you encounter issues regarding BLE connection.

closed 2022-01-31 by JF002

JosuGZ, 2023-10-04:
I'm experiencing something similar. I have the HR always on and at night the connection ends up dropping after about 2 hours. During the day it seems to work ok.

Once that happens, GadgetBridge is not able to reconnect (but a custom app I have is eventually able to reconnect, after one or two errors), then, GadgetBridge is able to connect again (whithout restarting the watch, although somethimes I disable and reenable the bluetooth on the settings).

I'm using 1.13 (it started happening with stock 1.13 but now I'm testing with a small change to prevent the hrs to sleep when the screen is off).

When I was using stock, I had the watch set up to wake with shake and wrist and everything + 30s of screen time so it would take as many measures as possible. I haven't yet been able to collect a whole night worth of hr data.

I have also tried Sleep as Android with the same results: drop after about 2 hours (in this case, only the modified firmware).

JosuGZ, 2023-10-04:
> I'm experiencing something similar. I have the HR always on and at night the connection ends up dropping after about 2 hours. During the day it seems to work ok.
>
> Once that happens, GadgetBridge is [...]

Oh, now that I think, maybe I can get a HCI log and see whats going on with WireShark.

JosuGZ, 2023-10-08:
> Oh, now that I think, maybe I can get a HCI log and see whats going on with WireShark.

Oh I'm pissed off >:( Android has not only made the HCI log featture harder to use, they have completely broken since it "forgets" old data. That means I don't get the 2-3am data that I need... Maybe trying to retrieve it first thing in the morning...

JosuGZ, 2023-10-28:
After a while, this never happens if I leave the phone charging, so the problem seems to be Android.