InfiniTimeOrg/InfiniTime #709: [Feature request] 2FA watchapp

adocampo, 2021-09-30:
Title says everything. I had plenty of TOTP-secured passwords, I activated OTP whenever I can. On Pebble I had a watchapp to see all them.

The OTP tokens were configured through the companion app, and once synced, watchapp was completely standalone as TOTP is a very simple protocol which you just need to have the proper time synced, and now that we have persistent time, I think it's the time to proposal to a curious developer to create a really cool and useful watchapp.

I guess it will be necessary a companion app as well, because entering a new OTP token is done by scanning a QR or entering quite a long hash with numbers and letters. The companion app should let the user order the list of entered tokens and also let the user delete them.

Once entered the OTP and synced to the watch, the watchapp should show a list of all the tokens and tapping on one of them, show the OTP code and the time it will live, like a descending progress bar or another animation, or a decreasing number. 

[image]

The screen should be awakened at least all the remaining time for that token's life, it would be pretty frustrating to lose the opportunity to enter the code because the screen suddenly dims, and every second counts.

What do you think?

arvchristos, 2021-09-30:
This would be a game changer for me too! I am using MFA and having TOTP codes on my wrist would be great. Though I am not sure if this would require a new companion app

tmilburn, 2021-10-01:
@adocampo  Can you remember which pebble watchapp you were using? 

The protocol is relatively straight forward there is some documentation at https://github.com/google/google-authenticator/wiki but the issue will be sharing the shared secret with the watch in a secure way. Currently the pinetime doesn't require pairing so bluetooth communication can be sniffed.

adocampo, 2021-10-01:
> @adocampo Can you remember which pebble watchapp you were using?
>
> The protocol is relatively straight forward there is some documentation at https://github.com/google/google-authenticator/wiki but [...]

Just plugged in my pebble... and it started! (it died on the beach this summer) Let me charge it a bit and I will take a look of the exact name of the watchapp.

I remember that companion app did something strange: you had to open it, then a message appeared saying it will close, and you had to re-open it again for "security reasons". So every time you opened the companion app to manage your tokens, you had to open it twice. I wonder now if that was the reason.

Avamander, 2021-10-01:
Duplicate of https://github.com/JF002/InfiniTime/issues/310

adocampo, 2021-10-01:
> @adocampo Can you remember which pebble watchapp you were using?
> 
Well, after a few minutes, my pebble died again and now it's on a bootloop, but looking in the Net I've found it.
It was this one: https://apps.rebble.io/en_US/application/52f1a4c3c4117252f9000bb8 which github is this https://github.com/cpfair/pTOTP, perhaps you can find some useful information
This is a color picture I also found, please note the blue bar at the bottom, this bar shows the remaining time for the selected token.
[image]

ajack2001my, 2021-10-01:
I believe this is best done once the watch can receive individual apps and watchfaces and not in it's current state where everything is put together in one update regardless if the user wants to use the app/watchface or not.  The question also comes to mind if GB 0.60.0 can do all this or if it's better to write a separate PineTime specific app. Perhaps port over the PineTime app from another platform?

Just my 2 cents on this topic.

Avamander, 2021-10-02:
Closing as a duplicate of https://github.com/JF002/InfiniTime/issues/310

closed 2021-10-02 by Avamander