InfiniTimeOrg/InfiniTime #710: [Enchancement] Reduce touch screen sentitivity

kieranc, 2021-10-01:
I find, and I don't think I'm alone, that the touch screen is extremely sensitive, probably since the new touch driver was merged. The new touch driver is clearly a huge step forward but I think it needs some simple limits putting in place to define a minimum length/distance of a swipe for example, or duration of a tap. I often find that I accidentally end up dragging down the notification shade when I try to tap a button, meaning that something with only a few pixels of movement was interpreted as a swipe rather than a tap, and I think it should behave differently.

Riksu9000, 2021-10-01:
Adding a minimum threshold is very easy, but I'm not sure it would make things better.

Due to the way LVGL handles touch, making swipes less sensitive would make clicks more sensitive, making it more likely for swipes to be recognized as clicks, and I don't think that would make things better overall.

The issue is that LVGL doesn't care if the touchpoint moved, it will always click on release anyway. In LVGL8 this would've clicked on the green button instead, which is better, but ideally it shouldn't activate a click on anything. #616

I find that clicks are too sensitive, but I think that's an issue with LVGL, and filtering the touch points before giving them to LVGL seems hacky and probably wouldn't work in all situations without bringing back different touch modes again.

https://user-images.githubusercontent.com/37774658/135605494-6aa3e7dc-025a-41bf-a1dc-8ccc181454a0.mp4

kieranc, 2021-10-01:
Thanks for your input, I know you have a much better grasp of this than me! Given that upgrading to LVGL8 is going to be very complicated, would it be worth patching or backporting the LVGL event handler to give us better behaviour instead of filtering the data we pass to LVGL?

Avamander, 2021-10-02:
I read the title as `Enchantment` and for a second I were very confused, it should probably be `Enhancement`.

ck-telecom, 2021-11-16:
the touch controllor should always updated(irq generated) when moving finger, or finger down and up, but do not how to set the hardare register

closed 2022-06-07 by kieranc