InfiniTimeOrg/InfiniTime #712: [Enhancement] Display a visual indicator for battery-intensive configuration options

Avamander, 2021-10-02:
It's difficult for users to know which features add additional battery drain. It would be nice to display a visual indicator next to those configuration options, :zap: for example.

Riksu9000, 2021-10-02:
Are there options like that? The only thing I can think of is that the touch screen isn't disabled if double tap wakeup is enabled. According to the datasheet rev 1.0 it should automatically go to a low power mode where it consumes less than 100uA, but rev 1.1 doesn't mention this. Can someone test if double tap wakeup increases power consumption considerably?

JF002, 2021-10-03:
Well, there are many features that use *more* power when they are enabled (raise to wake, shake to wake, constant HR monitoring, BLE connection, higher brightness levels, longer display timeouts). 
This could be a good idea, at the condition that this indicator does not clutter the UI and confuse the users :) I leave that to UI/UX experts !

tmilburn, 2021-10-04:
I personally prefer the UI which shows the user how much of the current battery has been consumed by different system level tasks. As an example:

75% battery used, 25% remaining
52% used by screen LED
7% standby power
6% bluetooth usage 
5% CPU running time
3% vibration motor
2% step counting

This would require timing how long each task is running and estimating the power usage (so won't be 100% accurately) but this gives enough detail that a user can change how they use the watch.

Avamander, 2021-10-04:
I think power monitoring is a separate topic, this is meant to be a visual indicator for those features that are known to cut battery life in half.

tmilburn, 2021-10-04:
> I think power monitoring is a separate topic, this is meant to be a visual indicator for those features that are known to cut battery life in half.

And which features are they? Only heart rate monitoring comes to mind. I think working it into the UI is difficult because it isn't a pattern users will expect so will require some text somewhere to explain what the icon means.

Avamander, 2021-10-04:
> And which features are they? Only heart rate monitoring comes to mind. I think working it into the UI is difficult because it isn't a pattern users will expect so will require some text somewhere to explain what the icon means.

I think a power icon is fairly self-explanatory. "Something to do with power" and they can look it up if they really wish to.

In addition to HR, there's tap to wake, shake to wake and all those features.