InfiniTimeOrg/InfiniTime #721: Coding style: Ban auto type?

Riksu9000, 2021-10-05:
Auto can save the developer two seconds of time not having to look up the return value or assign a properly sized integer type. The type is set by the compiler, so it's no different than specifying the type explicitly.

Having auto in code results in the reader or maintainer having to look around to figure out the actual type, when the original developer probably knew what it was, but didn't bother specifying it. It just seems lazy to me and I see no reason to use auto in embedded development.

There's a tidy warning that suggests using auto when casting to avoid duplicate type names. Using auto in this case is fine as the type has already been specified, but it's not helping much either, and it might be better to simply not use auto anywhere.

Avamander, 2021-10-05:
Switch to an editor that can tell you the type.

dyamon, 2021-10-05:
> Switch to an editor that can tell you the type, it's the easiest solution.

IMO, while there is some truth in this, code should be readable even if printed on the back of a cereal box. And, indeed, abusing `auto` is not great for clean readable code.

What about just keeping an eye out for this in future PR reviews? I would consider acceptable the use of `auto` when the type is clear **from the immediate context**, but suggest to change it to explicit typing when this is not the case.

Avamander, 2021-10-05:
Given the generally positive opinion of the lack of explicit types and readability of e.g. Python and the generally negative opinion of the verbosity of Java's typing, I don't think a ban is reasonable.

Especially for the sake of some editor being unable to tell the type to the user who doesn't know it, over the benefit of the type being not important in that context and improves readability for most.

geekbozu, 2021-10-05:
I know I tend to use auto when I am not sure of the best variable type to use yet. while assuming the compiler will pick it for me. Does that mean its the best practice...no. I also make a point to not use it for anything that is getting returned or used out of the function. Its one of those niceties we could get rid of but I feel just setting some guidelines around how to use might be better. 
I have no hard opinions on this either way as long as we come to a consistent solution :)

Riksu9000, 2021-10-10:
I'd still say that while auto here may be acceptable, at best it doesn't improve readability, and at worst it forces the reader to look around for the type, so I wouldn't use it anyway.
https://github.com/InfiniTimeOrg/InfiniTime/blob/90fdea29b80016c6d0e6d518853c9251555ea695/src/drivers/Hrs3300.cpp#L58

I guess my suggested rule could be extended to not using auto for basic data types and integer types (unless to avoid duplication?), since I only really see the benefits of auto when used with special types.

That way when I see auto, I can know it's some special type, which would improve readability if it was consistent. That way seeing auto means "You might not know this type, but it's probably fine", instead of "You might know this type, but I'm not telling you". This is what would make the most sense to me at least.

I personally like having rules, so there's less guesswork and the code base could be extremely consistent without surprises. This is why I don't enjoy having `and` `or` along with `||` `&&` as well. I don't know if I'm asking for too much though.

JF002, 2021-10-16:
>  I don't know if I'm asking for too much though

Of course not! I think your concerns are completely valid. We need to have a consistent codebase and every contributors should be able to read the code easily without expecting surprises on each and every line of code.

To me, the most important think is the readability of the code. If it's easy to read, it's easy to maintain and extend. But I also understand that everyone have their own habits and affinities.

So, let's try to write a rule/recommendation about the use of `auto` and boolean operators so we can it it to our coding guidelines :)

geekbozu, 2021-10-16:
I'm with JF, I think these questions are important. We should not harp on it so much to push people away from contributing but making sure the code is maintainable is just as important as getting stuff done. Clang helps a lot with some of it. There is still more we can do however!

> So, let's try to write a rule/recommendation about the use of `auto` and boolean operators so we can it it to our coding guidelines :)

Agreed!