InfiniTimeOrg/InfiniTime #725: [DFU]: restrict by SD-req field

ildar, 2021-10-07:
### 

- [X] I searched for similar bug reports and found none was relevant.

### What happened?

some users brick their devices by installing the DFU ZIP files of InfiniTime

### What should happen instead?

FW metadata in ZIP should disallow

### Reproduction steps

Currently DFU zip is produced without `--sd-req` option: https://github.com/InfiniTimeOrg/InfiniTime/blob/develop/src/CMakeLists.txt#L903
This makes dfu.zip harmful for some bootloaders used on nRF52 devices.

A simple solution here is to add `--sd-req` with pseudo SD code.

### Version

HEAD

fanoush, 2021-10-07:
just adding more info - nordic SDK11 bootloader or adafruit bootloader which is derived from it  contains following code
https://github.com/adafruit/Adafruit_nRF52_Bootloader/blob/master/src/dfu_init.c#L172
if the array inside dfu  zip is empty, the package is typically accepted and will brick the device

list of valid codes are e.g. here https://github.com/NordicSemiconductor/pc-nrfutil/blob/master/nordicsemi/__main__.py#L535 so some low value like 0x01 should work

there are people trying to switch to infinitime and brick devices while trying, see eg. first post of https://github.com/InfiniTimeOrg/InfiniTime/issues/62 or  https://github.com/InfiniTimeOrg/InfiniTime/issues/62#issuecomment-868407337

JF002, 2021-12-30:
> some users brick their devices by installing the DFU ZIP files of InfiniTime

What did those users expect by flashing a DFU from InfiniTime (that is intented to run on a PineTime flashed with [our specific bootloader](https://github.com/InfiniTimeOrg/pinetime-mcuboot-bootloader) on another device running another bootloader and software stack?

ildar, 2021-12-31:
InfiniTime firmware on PinTime with another bootloader, e.g. wasp-bl

fanoush, 2021-12-31:
> What did those users expect by flashing a DFU from InfiniTime ... ?

InfiniTime of course :-) Either for PineTime or other watches from same Moyoung manufacturer that makes Pinetime (e.g. P8, P22, .. ) they share mostly the same pinout and same components so Infiniime may run as is or with slight changes. Often those users are a bit clueless and/or blindly follow guide made by someone else (who is slightly less clueless). This is internet. They are also 'spoiled' by testing Arduino, Espruino and WaspOS DFU zip files in similar way without fully understanding how stuff works. Most other such zip packages prevent them to shoot themselves in the foot by having correct metadata inside.

> that is intented to run on a PineTime flashed with our specific bootloader

Despite having different bootloader and architecture  InfiniTime guys (ab)used Nordic legacy DFU zip format, so the package as it is made now can be flashed to any Nordic SDK11 based device and will brick it. That is possible because the package is too generic and not following Nordic guidelines (by claiming it is application that will run on top of any Nordic softdevice).

JF002, 2022-01-01:
As far as I understand, the goal of `--sd-req` is to ensure that a version of the SoftDevice compatible with the firmware is running on the target hardware before flashing a new version of this firmware. This allows application developers to upgrade the version of the SoftDevice while ensuring their users won't be able to flash a new version of the application (based on the new SD) on a device running an older (incompatible) version of the Softdevice. This is needed because the softdevice can be updated independently from the application.

However, it's not intended to be used to check that the user tries to install a firmware on a compatible hardware!

I mean, even if we implemented the proposed solution, nothing would prevent users from trying to flash a random DFU file found on the internet on any device running on a NRF MCU. 

In my opinion, this proposed change tries to fix the issue at the wrong place. This is the Internet and users will do anything, that's true. And we can provide the users with as many documentation, BIG RED WARNING SIGNS or update procedure, they'll still do anything... But trying to fix that issue in InfiniTime is wrong... What if InfiniTime was based on the NRF SoftDevice? Users would be able to flash InfiniTime on any other device based on the SD and it won't run either...
To me, this issue must be handled on the bootloader side : it should not blindly accept any piece of binary code that is provided by the user without some kind of failsafe mecanism. Alternatively, if there is a risk of bricking the device, users should be warned more than once so they know what they are doing.

> Either for PineTime or other watches from same Moyoung manufacturer that makes Pinetime (e.g. P8, P22, .. ) they share mostly the same pinout and same components so Infiniime may run as is or with slight changes.

Infinitime has never claimed to be compatible with other hardware than the PineTime, and, to me, trying to install InfiniTime on those other devices makes as much sense as trying to flash an Apple watch with a Samsung firmware...
The fact that those hardware are similar and sometimes compatible is pure luck... Also, there are many variants of the P8, and some of them are based on another touchpanel that is not compatible with InfiniTime, for example.

> the package as it is made now can be flashed to any Nordic SDK11 based device and **will brick it**.

I truly don't understand the part **in bold**... As far as I know, the Nordic bootloader is very reliable and safe, even safer than InfiniTime, as the bootloader implements the BLE stack and OTA functionality, so that a new firmware can be flashed even if the current firmware is completely trashed.

Even InfiniTime's bootloader won't likely brick if you OTA an incompatible firmware, as it simply won't run and the bootloader will automatically revert to a previously working firmware. 

> That is possible because the package is too generic and not following Nordic guidelines (by claiming it is application that will run on top of any Nordic softdevice).

You cannot assume that a firmware built for a specific version of the NRF SoftDevice will work on any device using that specific version of the SoftDevice. As I said before, there are many devices that are based on the NRF SoftDevice, and there's very few chances the firmware for a BLE thermometer will run on a P8 smartwatch, even if they run the same version of the NRF SD.

As a comparison, nothing prevent me from using diesel on my gasoline car... but it will break my car and I'll be the only one at fault.

> InfiniTime firmware on PinTime with another bootloader, e.g. wasp-bl

Is it a custom bootloader, or is it based on the NRF bootloader? If there's any issue with the switching procedure between InfiniTime and WaspOS on the PineTime, I'll be more than happy to work on it with Daniel.

fanoush, 2022-01-01:
> I truly don't understand the part **in bold**... As far as I know, the Nordic bootloader is very reliable and safe, even safer than InfiniTime, as the bootloader implements the BLE stack and OTA functionality, so that a new firmware can be flashed even if the current firmware is completely trashed.

The package as it is made  (without --sd-req) declares that it can run on *any* softdevice (which  is not true at all!). That means nordic DFU bootloader will happily flash it (SDK11 bootloader will flash binary data from the zip  to 0x1c000 address) and then will happily jump into it causing neverending boot loop. And then there is no way out of it. The firmware is marked as valid as the checksum inside zip is correct so on every reset bootloader will jump into it without giving you chance to enter DFU mode.

If Infinitime/mcuboot would use own DFU format instead of abusing nordic legacy dfu (despite not being based on nordic softdevices at all) or if it would at least use --sd-req with some special id the package would be less dangerous.

> You cannot assume that a firmware built for a specific version of the NRF SoftDevice will work on any device using that specific version of the SoftDevice. 

There is optional `--dev-revision` and ` --dev-type` that can be used for that but hardly any manufacturer uses it because Nordic  SDK example bootloader does not implement such checks. So almost all SDK11 based devices in the field will accept such generic package. I see InfiniTime uses 0x52 for --dev-type (which is generic `nrf52` value) and no --dev-revision so there seems to be no attempt to make the package specific to PineTime via this either.

And BTW most of Espruino packages that I maintain are truly generic and will run in every nrf52832 device under the sun that runs specific version of Nordic softdevice. The hardware specific 'drivers' are all written in javascript and uploaded later over BLE on top of generic build.  But anyway this is all unrelated to the issue of too generic infinitime dfu zips that claim to run on top of any softdevice.

fanoush, 2022-01-01:
> As far as I understand, the goal of --sd-req is to ensure that a version of the SoftDevice compatible with the firmware is running on the target hardware before flashing a new version of this [...]

Correct. And since you can hardly build softdevice-independent app not having `--sd-req` mandatory for building `--application` package type is IMO  bug in nrfutil/adafruit-nrfutil.  And I think this was fixed in later nrfutil versions implementing 'secure' nordic dfu format. Not specifying `--sd-req` make sense only for bootloader+softdevice dfu package type because that one will completely replace softdevice and bootloader so previous version does not matter.

> However, it's not intended to be used to check that the user tries to install a firmware on a compatible hardware!

No it is not. And nobody claims it is.  And yet the way InfiniTime uses it is wrong because unfortunately no `--sd-req` specified  means  *any* softdevice.  Which can be also considered as a bug or poor choice in nordic bootloader code (and derived [adafruit bootloader](https://github.com/adafruit/Adafruit_nRF52_Bootloader/blob/master/src/dfu_init.c#L177) with `DFU_SOFTDEVICE_ANY ` constant) but is a reality.

JF002, 2022-01-02:
You are giving intentions to the integration of the DFU protocol into InfiniTime it has never had... Interoperability with other projects/products based on NRF devices and NRF SDK has never been a goal. I decided to implement the DFU protocol because it's a tool that is simple that exists. You want to call that "abusing", fine, but I disagree... Why should I reinvent the wheel when what I need already exists? Again, the fact that other products implement the same protocol is cool, but this protocol is not meant to avoid users trying to flash the firmware of a product on another product (except for you project that doesn't use any hardware except the CPU...).

As you said, there are a lot of design flaws : the NRF bootloader that overwrites the memory without any fallback, the protocol that provides additional info (device type & revision) that no implementation actually use, nrfutil that allows inconsistent parameters to build the DFU file,... So I'm wondering... what should InfiniTime try to (and will fail to) fix all these design flaws?

But OK, I understandwe'll never agree so... let's end that here : We can add those parameters in the DFU files from InfiniTime so we'll prevent some of those issues. But I'll accept those changes only if we ensure that no companion app break with those changes.

Avamander, 2022-01-02:
I would add that the DFU protocol is quite irrelevant in this case because what should be checked by flashing software is the metadata in the DFU package itself. That metadata should be checked by whatever is writing the package.

Now if the only piece of software that should be checking isn't actually checking anything, it has the thought process of "oh we don't know for which device this package belongs to, but HEY at least it's SD-compatible, let's flash it" - then that software should probably be fixed, or throw a warning. There are also other components at play that would avoid this I won't dwell on right now.

It probably would be ok to add that metadata to the DPU package, but simply being able to pipe bytes over BLE doesn't mean it or one should. I will also take a look if I can lock down Gadgetbridge's uploader unless someone's absolutely certain, it only does a few basic checks right now.

fanoush, 2022-01-02:
> We can add those parameters in the DFU files from InfiniTime so we'll prevent some of those issues. But I'll accept those changes only if we ensure that no companion app break with those changes.

Oh, great, that would be nice.

When checking where it is used https://github.com/InfiniTimeOrg/InfiniTime/search?q=adafruit-nrfutil it is not that much.
As for what value can be used - it is 16 bit value and when not specified via `--sd-req` adafruit-nrfutil puts 0xFFFE there which means `DFU_SOFTDEVICE_ANY`. Nordic uses nonzero values below 255 so `--sd-req 0` or `--sd-req 0xFFFF` would do the trick. 

Just to clarify again why it makes sense - there were several cases like this one https://github.com/InfiniTimeOrg/InfiniTime/issues/62#issuecomment-868407337 where someone was trying to install InfiniTime but got the order of dfu packages wrong or missed one between and got a brick. This can happen again and is easily avoidable by making those zips harder to mix up.