InfiniTimeOrg/InfiniTime #748: Evolution of the documentation

JF002, 2021-10-16:
The documentation (for end-users and for developers) is not well maintained for now. The doc is scattered across multiple pages and places, is probably out of date and not visible enough for the end-users.
I think it would be a good idea to try to update and improve the doc to improve the users and developers experience!

Now, the question is : do we keep the documentation (for developers and/or end-users) in the InfiniTime project or do we host it somewhere else?

- In another git repo (could be hosted in the InfiniTime organization, of course)
- In Github wiki/pages
- On InfiniTime' website (https://infinitime.io), which I unfortunately do not maintain quite well :/
- anything else?

Splitting the doc from the code would allow to maintain it more easily and to delegate the maintenance to some contributors. But it would have the disadvantage to de-correlate the documentation from versions of InfiniTime. Can we handle that?

So, I open this issue to talk about the documentation : how would you manage the documenation of the project?

ntindle, 2021-10-16:
GitHub pages works well for the arcane algorithm archive. It’s better than wiki imo because it supports more features and can be setup to use markdown as well. 

TL;DR; +1 GitHub pages

geekbozu, 2021-10-16:
I'm leaning towards GitHub pages as well. its a nice middle ground between all the options...oh and we do not have to maintain the back end...GitHub does that for us ;)

ncartron, 2021-10-18:
agreed with the GutHub, that makes maintaining the doc easier (not sure how this would be done with infinitime.io?) - and of course hosting is done by GH

ncartron, 2021-10-18:
@geekbozu agreed, docs.infinitime.io (or something similar) would be nice

MrMinimal, 2021-11-04:
@JF002 Regarding your GitHub pages question:

You can think of it as a directory. It is a seperate repository, but GitHub's servers only host the contents of the main branch - so it's like having a single directory to store static files in.

MrMinimal, 2021-11-04:
## Proposal
I want to make a case for **keeping the developer and user documentation in the same repository**.
The main reason for keeping developer documentation in the same place as code is to minimize them being out of sync.

But user documentation shouldn't be out of sync either - more crucially, the less technically versed users might even be more impacted by outdated documentation.

If your main motivation to split documentation and code into seperate repositories is maintainer workload, you could assign "documentation maintainers" that can only merge doc related PRs to offload the work from code maintainers.

I agree that user documentation is updated more frequently than the software is released, one could make the case for just keeping the docs in the same state as the "development" branch is.

## Hosting options
1. **GitHub pages** - it's free, it's simple but you cannot have multiple documentation version (e.g. per branch). Pretty good SEO as far as I know
2. **ReadTheDocs** - it's free, it can host multiple versions of documentation (support of stable versions possible as well as development versions), Pretty good SEO, supports having docs.infinitime.io pointed to it.
3. **infinitime.io** - Maintaining a webserver takes effort and is probably not free either

JF002, 2021-11-07:
Thanks for your feedback so far!

## Where do we store the doc sources ?
From your comment, it looks like storing the whole doc (end-user + developer) in the same repo than the code would be the best.
I think we could create a team of contributors more focused on to the documentation so that the workload is shared across many contributors.

## Where do we host the docs?
Now, let's assume that the doc is stored in `/doc` in the repo as .md files (+ pictures and other static files). How do we transform them into nice HTML files and send them to the hosting platform?

If I understand correctly:
 * **GitHub Pages** does that for us
 * **ReadTheDocs** should also do that for us. It's more feature complete (like support of multiple versions), it's open source friendly but displays an ad.
 * We can also selfhost a hosting platform on the server that hosts **infinitime.io** (I rent the server at Hetzner for private use, so there will be no additional cost). It should even be possible to selfhost ReadTheDocs but it'll require a bit of setup and maintenance time.

I doubt I'll personally be able to allocate much time to the maintenance of our infra, but is someone(s) are willing to help maintaining infinitime.io and the doc platform, we can consider it. Otherwise, let's just use ReadTheDocs. As I said, I have full control on infinitime.io, so we'll be able to setup a subdomain (doc.infinitime.io) to whatever hosting platform we choose !

## Next steps
I think the next steps is to define the structure of the documentations, maybe a table of contents, for example.
But before going further, I would like to ensure everyone is ok with those decisions:
 * we store the documentation sources on the Infinitime repo
 * we use ReadTheDocs as a first hosting platform

# Inspirations
Here are some projects that have, IMO, good documentation:
 * [LVGL](https://docs.lvgl.io/master/index.html) : LVGl is the UI library we use in InfiniTime. The doc is easy to browse and I can easily find info and examples when I need them. They probably selfhost it using ReadTheDocs theme (`Built with Sphinx using a theme provided by Read the Docs.` on the bottom of the page)
 * [Meshtastic](https://meshtastic.org/docs/getting-started) : Meshtastic is an ecosystem that provides peer-to-peer text communication based on LoRa protocol. I don't know what documentation processor and hosting platform they use, but I really like the structure of the docs:
     - Getting started : quick introduction to new users
     - Software/Hardware : more detailed info about the project for the end-users
     - Developers : how to build the projects, API documentation, architecture,...
  The docs are not "multi-versionned" so I guess you're supposed to be using the last version anyway ;)

ntindle, 2021-11-07:
I agree that those are the best ways forward for the project in its current state. 

GitHub Pages is excellent for things that are not released to users with versions but in this case, read the docs is better. 

You have a +1 to move forward with this from me

ncartron, 2021-11-11:
Thanks for the sum up @JF002 , and thanks everyone for your comments.

Regarding:
>   we store the documentation sources on the Infinitime repo
>   we use ReadTheDocs as a first hosting platform

That works for me, I will start working on a structure/table of content over the week-end and will update this issue once I have something I like.

JF002, 2021-12-30:
I'm closing this discussion as work has already begun in #838

ntindle, 2021-12-30:
One last question worth considering:

how do we keep docs up to date? 

Some ideas are:
- making sure all PRs have a docs reviewer or similar required to approve
- having all committers acknowledge that their docs are updated when opening a PR

JF002, 2021-12-30:
Yes, I would like to be a bit more rigorous regarding the maintenance of the documentation.
@ncartron also joined the team to help us on that topic (among other things, of course!).

ncartron, 2021-12-30:
Agreed.
This could be part of the PR template - like a checkbox «documentation updated / reviewed» or something like that.

JF002, 2021-12-30:
That's a good idea, @ncartron !

ncartron, 2021-12-30:
I believe they do - I can have a look and see what other projects are doing.
I know for instance PowerDNS does that, and its pretty useful to check for basic requirements, see screenshot:

[image]

ncartron, 2021-12-30:
this page explains how to create template for PR: https://docs.github.com/en/communities/using-templates-to-encourage-useful-issues-and-pull-requests/creating-a-pull-request-template-for-your-repository

JF002, 2022-01-01:
I re-opened this issue, as the discussion is continuing ;-)

ProgramminCat, 2023-09-05:
+1 for docs.infinitime.io (or something similar)