InfiniTimeOrg/InfiniTime #755: Add a way to make customized watch faces.

alphamodh0, 2021-10-17:
Thanks for your work, I'm exciting with Pine64's proyects. Can U add sny way to add custom watchfaces from the pair app? And a way to share this watchfaces with community.

Danimations, 2022-03-31:
Glad to hear there's some progress being made on this front. Please keep in mind the prospect of importing a 240x240px .BMP via Bluetooth and using it as a watch background.

Avamander, 2022-03-31:
Related to https://github.com/InfiniTimeOrg/InfiniTime/issues/321

Danimations, 2022-03-31:
What do you anticipate will be the best file format and specifications for an image?

JF002, 2022-08-02:
I've just created a more generic feature request about dynamic app and custom watchface to group multiple requests around that topic. Feel free to continue the discussion in #1262 ;-)

closed 2022-08-02 by JF002