InfiniTimeOrg/InfiniTime #804: PineTime keeps shutting down around 40% of battery

ajack2001my, 2021-11-01:
### 

- [X] I searched for similar bug reports and found none was relevant.

### What happened?

Checked my batter status, shows 51% and an hour later my PineTime is dead due to a flat battery. Happened a few times already.

### What should happen instead?

PineTime dies only when battery hits 0%

### Reproduction steps

Charge till full, use like normal and battery dies around the 40% mark. Suspect battery statistics is screwed up but need somebody who know the internal workings of the PineTime to verify.

### Version

InfiniTime v1.6.0 

### Companion app

GadgetBridge v0.61.0

lectrician1, 2021-11-01:
Battery readings are definitely screwed up. I have repeatedly gone to to bed with 48% battery and woken up with it dead.

Self-Perfection, 2021-11-16:
Something changes at around 40% of charge and rate of discharge increases ~10x times. <details><summary>Here is typical screenshot of charge graph:</summary>
[image]
</details>

escoand, 2021-11-16:
I think this belongs to #858.

ajack2001my, 2021-11-17:
With reference to issue https://github.com/InfiniTimeOrg/InfiniTime/pull/585,  can the watch be made to go into DumbWatch (switch off bluetooth, all internal services, and only display time/date) mode when battery is less than 10%. This allows the battery to last for the rest of the day until the user can get home and charge the watch. Instead of being a dead watch for the rest of the day. 

Please adjust the 10% to something more appropriate to ensure it'll last at least 12 hours before going flat...

escoand, 2021-11-17:
This would be #659

dadall, 2022-01-10:
Got this problem few days ago : the battery went from 46% at 8pm to 0% at 5pm for no reason.

milochristiansen, 2022-05-07:
I had this issue today. I charged my pinetime to 100%, it dropped over 20% on the first day of use, the used 5-10% per day until it hit ~40%. I then charged it fof a few minutes (reported 68%), and died without warning that evening. When I put it on the charger at home a couple of minutes ago I realized I had never validated the update from 1.6 to 1.9, so I had to update the firmware again (oops, my bad). After this update 1.9 happily reported 37% battery. This is impossible, it was on the charger less than 5 minutes. Clearly whatever is suposed to be tracking battery charge is unreliable at best,

JF002, 2022-05-07:
The PineTime does not have any hardware dedicated to the battery monitoring, so the battery level is monitored by measuring the battery voltage : 4.2v = 100%, 3.2v = 0%. This is a very inaccurate way to measure the battery level since voltage vary a lot with the power consumption of the pinetime (for example, the level measured will be higher when the pinetime is sleeping than when the display is ON and the CPU under heavy usage), making the voltage measurement very noisy. Also, the voltage measured when the pinetime is on the charger will be higher so InfiniTime will report a level higher than the actual level of the battery.

Also, the discharge curve of the battery is not linear : it discharges quickly between 100 and 80%, then slower until 40% and very quickly from 40%.

We've already done some work trying to take into account this discharge curve, but the last segment (40 -> 0%) is not handled quite well for now.

Albirew, 2022-05-08:
I made a screenshot of discharge bar of my pinetime while idle
looking at the graph, i believe a logarithmic formula may needs to be applied to the data to flatten the curve

[image]

milochristiansen, 2022-05-09:
That looks a lot like a standard battery discharge voltage curve. Does Infinitime not correct for the battery discharge curve *at all?!?*

escoand, 2022-05-09:
See #719 and maybe more

wmertens, 2022-09-14:
I'm encountering the same (new owner since a week). Battery seems fine, next morning it's dead (probably just after Sleep as Android uses the haptics to wake me up).

I see that #719 was merged into the develop branch, does that mean it should be in 1.10? Because if so, it's not working :-(

JF002, 2022-09-17:
#719 improved battery level measurements, but didn't fix that huge drop at 40%. However, #585 might fix it.

fladrif, 2022-09-22:
Seeing the same issues with @wmertens. On the other hand though, is there a new issue with a higher consumption of power? Currently taking in the 40% cliff into account, the battery will currently last only 5-6 days, whereas it seems from other commenters that it used to last 7+ days. 

Thanks for all the work on this btw, everything else (plus updates) have been amazing.

strayobject, 2022-09-26:
@fladrif 4-5 days is normal from my experience. I can stretch it to 8 days, but it all depends on how active I am, screen brightness and if the heart beat monitor is on/off. In general you should not be running the battery down and charging all the way up anyway.

Riksu9000, 2023-01-30:
Fixed by #1444 in the next release.

closed 2023-01-30 by Riksu9000