InfiniTimeOrg/InfiniTime #807: Failed to find via Bluetooth

zabbal, 2021-11-04:
Infinitime 1.2.0 is not visible while scanning for Bluetooth - tried several phones with Gadgetbridge and NRFConnect.

Could it be hw issue preventing OTA?

Avamander, 2021-11-04:
Reboot both your phone and InfiniTime, delete it from under known devices in Bluetooth settings, do not pair when adding to GB.

zabbal, 2021-11-04:
Rebooted both, but the problem is that there's nothing under "known devices": I've never managed to get infinitime visible to add it there in a first place

JF002, 2021-11-04:
A few bugs were fixed in 1.5 and 1.6 that improved the reliability of BLE connectivity.
As you're running 1.2, it's possible that your phone have some difficulties to connect to your PineTime. However, rebooting the watch (long press on the button until reboot) should work.

Could you check [this comment](https://github.com/InfiniTimeOrg/InfiniTime/issues/302#issuecomment-944953772) where I give some tips to improve the discovery and connection of the watch?
Also, please ensure that **only one** application and **only one** device try to scan an connect to your watch. InfiniTime does not support multiple connections at the same time.

zabbal, 2021-11-05:
Indeed, the trick was connecting right after rebooting the watch - after that update went according to instructions.

closed 2021-11-05 by zabbal

Obikawa, 2023-06-12:
Please forgive me for posting to this old discussion.
I consider this less offenive than opening a new issue just to ask two simple questions.

> Could you check [this comment](https://github.com/InfiniTimeOrg/InfiniTime/issues/302#issuecomment-944953772) where I give some tips to improve the discovery and connection of the watch? Also, [...]

How hard is the connection limit?
Should I abandon my hope for the ability to pair two phones?

After using the PineTime for two days I already love the ability to receive all notifications with a quiet vibration on my arm.