InfiniTimeOrg/InfiniTime #812: incoming phone call from gadgetbridge continuously vibrates

clach04, 2021-11-06:
### 

- [X] I searched for similar bug reports and found none was relevant.

### What happened?

incoming call answered on phone (and hung up) infinitime 1.60 continues to vibrate

### What should happen instead?

should stop vibration when answeerd/hung uo

### Reproduction steps

* call phone
* answer phone, e.g. on handset, car, headphones, etc.
* optional hang up

Watch continues to vibrate

### More details?

Potentially dangerous when driving as dismissing requires looking at watch and taking eyes off road.

### Version

1.60

### Companion app

gadget bridge

escoand, 2021-11-06:
Duplicate of #367

geekbozu, 2021-11-07:
Duplicate closing,
Thanks guys :)

closed 2021-11-07 by geekbozu