InfiniTimeOrg/InfiniTime #819: The letter "I" in top left corner

rsheink, 2021-11-10:
Hi all,

From time to time I get the letter "I" it top left corner. Here is a screenshot:

[image]

What does this means?

Riksu9000, 2021-11-10:
This icon shows that you have unread notifications. It will show up if you have your watch set to silent mode in quick settings (bell icon) and you have received a notification.

I don't think this is documented, so maybe this can be considered a documentation issue.

rsheink, 2021-11-10:
Oh, I understand. Thank you very much.

SteveAmor, 2021-11-13:
For consistency I raised #776 to move the i to the top left on the analogue watch face.

drzacek, 2021-11-24:
It would be nice to add "i" to the notification itself. 
We got 5 slots and no way to tell when they came.

Riksu9000, 2021-12-04:
It is now documented what the icon means.
https://github.com/InfiniTimeOrg/InfiniTime/blob/develop/doc/gettingStarted/gettingStarted-1.0.md#digital-watch-face

closed 2021-12-04 by Riksu9000

moriel5, 2021-12-16:
I personally think that it would be nice to have several icons, in accordance to the type on notification (if possible), like missed call, text, or miscellaneous.