InfiniTimeOrg/InfiniTime #822: Waking up sometimes shows a minor display glitch

Riksu9000, 2021-11-11:
Doubletap to sleep and wakeup with a tilt soon after with precise timing. Sometimes this can happen when just checking the watch and sometimes it doesn't happen at all.

This probably started happening after #586, but it could also be that this just couldn't be seen when the wakeup was much slower.

Slow motion video showing the error.

https://user-images.githubusercontent.com/37774658/141324342-6ef95953-8672-4045-89d9-fb99a6122c2e.mp4

Seems like submitting a GitHub issue template/form isn't working. Trying a blank issue.

bakeromso, 2021-11-18:
I'm  on 1.7.1, wake on double tap and button (so not on tilt) and noticed a similar glitch when waking up. However, I'm unable to reproduce it. Leaving this here to inform that the issue might not be tilt related. Or were you @Riksu9000 able to reproduce it only for tilt specifically? If that's the case, then maybe ignore this comment.

mark9064, 2025-06-28:
I think this has been fixed somewhere along the line - if anyone can still repro this open a new issue

closed 2025-06-28 by mark9064