InfiniTimeOrg/InfiniTime #831: Update 1.6.0 to 1.7.0 fails

InFerYes, 2021-11-15:
### What Happened?

The update starts, but fails with "Error".

### What should happen instead?

The device should update from 1.6.0 to 1.7.0

### Reproduction Steps

Using GadgetBridge:

1. Download pinetime-mcuboot-app-dfu-1.7.0.zip
2. Open the file with GadgetBridge
3. The update process in GadgetBridge starts, GadgetBridge appears to be sending data but waits for the PineTime to receive
4. The PineTime shows `Firmware update 0%`
5. After a short while the PineTime shows `Firmware update error`

Using Siglo:

1. Open Siglo and select Automatic under "Firmware update"
2. Select tag "1.7.0"
3. Select asset "pinetime-mcuboot-app-dfu-1.7.0.zip"
4. The update process in Siglo starts, Siglo appears to be sending data but waits for the PineTime to receive
5. The PineTime shows `Firmware update 0%`
5. After a short while the PineTime shows `Firmware update error`3

### More Details

I have tested the update process on both Android 11 and PinePhone Arch Linux.

### Version

1.6.0
### Companion App

Siglo 0.9.5-1, GadgetBridge 0.59.3, GadgetBridge 0.62.0

Avamander, 2021-11-15:
Did you restart both sides in the equation?

InFerYes, 2021-11-15:
The PineTime has been restarted many times, and GadgetBridge and Siglo run on different devices/platforms.

Avamander, 2021-11-15:
Yeah but do try restarting the other side as well, this switching between the two might've left something lingering.

InFerYes, 2021-11-15:
I have both started the Android phone and PinePhone cold and rebooted, and I have restarted the PineTime inbetween attempts.

TechnologyClassroom, 2021-11-15:
I could not reproduce.  Updating with GadgetBridge worked for me on the first try.

InFerYes, 2021-11-15:
I have always done it with GadgetBridge, but I just can't get it to work. I have restarted both devices a tens of times, the update process just shows the red `Error!`. Downgrading gives the same error.

I tried Siglo because it GadgetBridge wasn't working for the update, but to me it rules out the problem lies with GadgetBridge (or Siglo). 

The PineTime still receives time updates and notifications via GadgetBridge. 

What can I try?

TechnologyClassroom, 2021-11-16:
When GadgetBridge acts up with me, I have found that deleting the device and then adding it again sometimes helps.  Not sure if it would help in this case, but it is worth a shot.

InFerYes, 2021-11-16:
That's standard modus operandi for GadgetBridge though. I have done it numerous times.

geekbozu, 2021-11-16:
This is typically just bluetooth dropping out or something equally bad. You can try doing it from the recovery firmware (hold button until red pinecone) or just keep trying until it goes though.

InFerYes, 2021-11-16:
When the pine is red, the PineTime isn't discoverable over BT. If I release the power button, the device will restart to InfiniTime.

TechnologyClassroom, 2021-11-16:
When I was first flashing OTA to 1.0, I had several failures.  My perceived issue was that the phone and watch were too close by being next to each other.  When I increased the distance to a few feet, I was able to get consistent transfers.  Worth a shot.

InFerYes, 2021-11-16:
No, it keeps saying `Firmware update Error!`. Keeping some distance between the devices and trying it in different rooms did not work.

Avamander, 2021-11-16:
Hold it until it turns blue, that restores the previous firmware version.

balacij, 2021-11-16:
I've got the exact same problem (with very similar conditions and attempted mitigations, it appears). I haven't tried it using Siglo (no PinePhone unfortunately), but I did try NRF Connect in addition to GadgetBridge. With mine, I get the same error but, at times, it will start the transfer, go up to 99% or even 100% and _then_ "Error!". Are there any logs generated that we might be able to use to understand why it's erroring out?

InFerYes, 2021-11-17:
My devices shows `Error!` _before_ GadgetBridge shows the "Upload in progress" screen, which says `0% at 0.00kbps (average 0.00kbps)`. When GadgetBridge shows the "progress", the PineTime goes back from the error message back to the watch face.

Avamander, 2021-11-17:
You can try redownloading the DFU file and you can also post Gadgetbridge logs from `adb logcat`.

InFerYes, 2021-11-18:
How do you "adb logcat"?

Arsen6331, 2021-11-23:
> I don't think GadgetBridge or Siglo are the issue, but rather the PineTime/InfiniTime 1.6.0

I haven't had any such issues with 1.6.0 itself. I have, however, had this "Error!" issue with the companion that I wrote. In my experience, this usually happens when InfiniTime doesn't receive any DFU packets for some time and times out. It turns out my companion was missing notifications from InfiniTime due to the speed at which they were being sent, and I fixed it by buffering the notifications. The symptoms were quite strange. It worked for a while, then stopped working, and worked very rarely from there, until I fixed it. That may not be it, especially if it's happening in multiple companions, but this did also happen to me in Siglo and Amazfish.

KawaiDesu, 2021-11-24:
Had the same issue trying to upgrade 1.7.0 -> 1.7.1. Errors was at 99% and 0% upload. But in the end I managed to upgrade firmware:
- Remove PineTime from Gadgetbridge
- Reboot phone (better turn off, wait a little and turn on)
- Reboot PineTime
- Optionally disable `CompanionDevice Pairing` in Gadgetbridge settings (it seems to causing problems in another case on another phone with Android 9)
- Add PineTime in Gadgetbridge (optionally without pairing: also possibly caused problems on metioned Android 9 device)
- Run firmware upgrade

In my case I have successfully upgraded with `CompanionDevice` and pairing.

TechnologyClassroom, 2021-11-25:
I was able to update from 1.7.0 to [1.7.1](https://github.com/InfiniTimeOrg/InfiniTime/releases/download/1.7.1/pinetime-mcuboot-app-dfu-1.7.1.zip) using GadgetBridge without issue.

JF002, 2021-11-28:
OTA sometimes fails in a similar fashion on my setup. I'm not sure, but I had the impression that unplugging both my phone and the pinetime from their respective chargers would help.

A few background info:
The companion app has the possibility to wait for a notification from the pinetime every 10 packets (200 bytes) to ensure everything is alright. This slows down the data transfer, but allows to detect errors sooner (in case of error, the transfer will stop with an error at any % between 0 and 99%).
If the companion app does not wait for the notification, transfer will be a bit faster, but the error will only be detected at the end of the transfer (99%).
If you see an error at 100%, it means that the data was corrupted somewhere between the companion app and the SPI flash of the pinetime and that the CRC is invalid.

InFerYes, 2021-12-01:
I have updated to 1.7.1 a few days ago. I simply downloaded the 1.7.1 zip and used the regular GadgetBridge method to update the device. I did nothing special, it just worked. 

I had tried everything for 1.7.0 but it always failed, 1.7.1 worked flawlessly from the first try.

Avamander, 2021-12-01:
Closing as you managed to fix it.

closed 2021-12-01 by Avamander

dascwo, 2021-12-02:
I'm having this same problem, but I can't use GadgetBridge because I don't have any Android devices. I've been trying to update the firmware using Siglo from three different devices (a desktop and a laptop both running Gentoo, and a PinePhone running postmarketOS), but I haven't had any luck. In my case, the watch never gets past 0% before the `Error!`. I was able to enable verbose mode with a source change in Siglo, and this is the output I'm seeing:

```
Start scanning
Found c7:ea:a7:ae:35:d3
File set!
[c7:ea:a7:ae:35:d3] Connected
self.mac c7:ea:a7:ae:35:d3
[INFO ] GNOME power settings not found, you're on your own for system sleep
preparing pinetime-mcuboot-app-image-1.7.1.bin for c7:ea:a7:ae:35:d3
Binary image size: 365704
Binary CRC32: 481291231
[c7:ea:a7:ae:35:d3] Resolved services
[INFO ] Enabling notifications for Control Point Characteristic
Notification Enable succeeded for Control Point Characteristic
[INFO ] Sending ('Start DFU' (0x01), 'Application' (0x04)) to DFU Control Point
Characteristic value was written successfully for Control Point Characteristic
[INFO ] Sending Image size to the DFU Packet characteristic
[INFO ] Waiting for Image Size notification
Characteristic value was written successfully for Packet Characteristic
```

I've also been tracing the Bluetooth packets in Wireshark and I see the "Start DFU" message (`0104`) succeed and the image size (`000000000000000088940500`) get sent, but I'm not seeing anything after that. I think that means that the problem is happening somewhere between the image size being sent and the notification I should get back from the watch that it was received?

I also tried hacking Siglo to re-send the image size after 5 seconds of no reply in case that was getting dropped, and hacking it to skip ahead to sending the "INIT DFU" message (`0200`) after five seconds in case the response notification is getting dropped. I'm not getting anything else back from the watch with either of these hacks, though.

What else can I try? @Arsen6331, can you point me to your companion app so I can see if the notification buffering change you made fixes this for me?

ITCactus, 2021-12-29:
sorry for posting to closed issue, but maybe my experience will help somehow ...
i had no issues when i tried to install 1.7.1 for the first time. and no issue when i played with new fonts for PR #887 using Debug build.
but Gadgetbridge mostly fails for Release build. i thought that was an issue with my build, but it can fails a lot with a officially released build, but again, it works ok with InfiniTime Debug build.
so, for me the "nRF Connect" app helps a lot with updating firmware on PineTime, as it can install the Release build in most times (but can also fails).
one thing to keep in mind - every time you see "Firmware update error!" on PineTime, you have to reboot it: even "nRF Connect" will not be able to install the firmware without it.

JF002, 2021-12-30:
I discovered an issue with Gadgetbridge recently (https://codeberg.org/Freeyourgadget/Gadgetbridge/issues/2527) : it subscribes to the raw values of the motion sensor as soon as it's connected. This triggers the update and the notification of those data over BLE. It consume CPU power, BLE bandwidth, and it seems to confuse the DFU service.

Gadgetbridge has already remove this subscription as the raw values were not used, and I noticed that it improved the OTA from Gadgetbridge a lot.

This fix will be released in the next version of Gadgetbridge. In the meantime, using NRFConnect or any other companion app should help.

ITCactus, 2021-12-31:
i can confirm, that thanks to the fix the [2527 issue](https://codeberg.org/Freeyourgadget/Gadgetbridge/issues/2527) in Gadgetbridge, with the 0.64.0 version InfiniTime firmware update works reliable (at least for me) - 5 successful installs from 5 tries :)

InfocuspSamyakVora, 2023-03-29:
I was trying to install custom stable firmware but I am getting firmware update error and now I couldn't find the watch and connect it with android applications - Gadgetbridge, NRF connect, BLE scanner.
Also tried connecting from multiple android devices and an iphone.

We have tried a few things for solving this issue:
1. Restarting the mobile as well as watch multiple times
2. Rolling back watch’s software version to a previously validated version
3. Checked out related github issues here but nothing worked for me
4. Tried completely discharging battery and starting fresh.

Now I am blocked by this issue, can neither update the firmware nor connect it with my watch.
cc: @JF002 

[image]

Avamander, 2023-03-29:
Kindly don't directly ping people unless in direct conversation.

Please open a discussion thread to diagnose your problem, it's not related ot this issue.