InfiniTimeOrg/InfiniTime #836: Torch settings

medeyko, 2021-11-18:
It's great that the torch is now ajustable, thanks!
But personally for me it has became much less usable due to the dim initial state.

I often use it to routinely light a room. For this I need a torch that lights at maximum level with minimum gestures.

The old version (firmware 1.6.0) required only TWO gestures: 1. swipe left for menu; 2. click on the torch icon.
Now (1.7.1) I have to make FIVE gestures: 1. swipe left for menu; 2. click on the torch icon; 3. swipe left to increase light; 4. swipe right to increase light; 5. click to turn the torch on. That's really a lot!

This is quite inconvient for my use case. I would like to either have an opportuninty to set the default light level to maximum and the torch state to on, or have some other short sequence (of no more than two gestures) that allows to turn the torch on maximum (for example, long button press or long click on the torch icon in the menu).

Also it's a pity that requests to support RED color were rejected. There are lot cases when low-energy photons are preferable, like photograhy or astronomy. Also, the red color is used to draw attention. Namely, I would use red torch when walking by roadside. It might be just a fourth "light level" option, a red one after three white ones (in the torch configuration or in the torch app itself).

For the latter use case, it would be nice to have a strobe light option (it draws more attention and conserves battery power), but probably it's another topic for discussion.

Riksu9000, 2021-11-18:
The default brightness is based on the backlight level, but maybe it makes more sense to default it to the highest level as that's what is most likely to be used.

The reason the flashlight is off by default is to not shine bright light in the users face when they launch it. The idea is that you need to look at the watch to open the app, but when the app is open, you can point it away and tap the screen to turn it on.

The long press idea is interesting. We could use that for something if we need to, but for now I think we can get by with just a bit of tweaking.

Red wasn't necessarily rejected. Red on the PineTime is very dim and it's not very pure I think, so I'm waiting for someone to actually use red light on the PineTime and tell me it's useful. Drawing attention when walking in the dark may be a good idea, but maybe not enough to justify having red when you can just use white.

medeyko, 2021-11-18:
Just a small comment, why I'd prefer a red light for the roadside walks: a red color is used to denote edge of the same-side roadside (to the right side for most countries, including my one). The white color is used to denote opposite roadside/opposite direction lanes (to the left side). So using white color for objects walking on the same-side roadside might be a bit misleading for the drivers. Also red color is less frequent than white one, and it is linked to drawing attention. Despite of course white light is still much better than nothing!
I know that red color is darker than white one, but it also means that it needs 3 times less power (because the PineTime's screen is OLED), as far as I understand.

medeyko, 2021-11-18:
> watch to open the app, but when the app is open, you can point it away and tap the screen to turn it on.
Ah, yes, so one can treat this as a long double click gesture! :)
Ok, then, if the default brightness level is max, seems it would be almost the same user experience as earlier (in 1.6.0) for my main use case.

As for the long button press - I think that shorcuts like long button press/double button press, right swipe/click/double click/triple click on main watch face might be quite useful. It may be handy to have an opportunity to link them to most used apps, but probably I should open another issue for discussing this (a bit later, after I exprore the issue deeper, including how difficult is to code this feature).

drzacek, 2021-11-24:
Maybe make it remember last setting instead of defaulting it to anything?

adocampo, 2021-12-15:
I use the torch app mainly when my baby cry at night, so now is quite more unpleasant to operate the torchlight than on 1.6.0. I would also like to open it with two taps or with quick access like a long press or with a swipe from the home screen (let's say to the right, where is no option right now, and could be used to place your "preferred app", for instance)

Shatur, 2021-12-27:
> The reason the flashlight is off by default is to not shine bright light in the users face when they launch it. The idea is that you need to look at the watch to open the app, but when the app is open, you can point it away and tap the screen to turn it on.

I think users can point it away before the opening the app by remembering the finger position. In both cases it will require less actions.

medeyko, 2022-02-05:
Please please please introduce a way to turn on the torch at maximum brightness level by a few gestures.

I use it every day, and still feel annoyed every time how many precise gestures I have to do for this.
(1.6.0 torch version was not annoying, but still is not ideal. But 1.7.0/1.8.0 version is really, really annoying.)

Riksu9000, 2022-04-14:
The middle brightness setting may not be very useful, so perhaps we should remove it and make full brightness the default option.

medeyko, 2022-04-14:
Yes, full brightness and on-state defaults are **really needed**.
The current version has a horrible UX.
I stopped to use the torch, because the need to make a lot of gestures was becoming more and more annoying each time.

There's even no need in off-state at all: one can block light by the hand one uses to  run the torch app and adjust the brightness not directly looking on the watch - the peripheral vision is enough for this. And when you close the torch app, there's no more light. So I don't see any inconvience if the on/off swich is removed.

Shatur, 2022-04-14:
I would go with "remove brightness control" and "on" by default.

drzacek, 2022-04-14:
If my voice counts, I also would say default on with no brithness settings. 

Personally, I don't even think that at full it makes that much light, so setting to anything less is meh.

medeyko, 2022-04-14:
> the brightness control not make much sense if it's going to shine bright light at first anyway. 
No, you are wrong, it still makes sense. I wrote it above, but let's write in more details.

I read about two cases, when people needed this control.

1. One wishes to be able to dim the torch when stepping into a room with a resting child, to be sure not to bother him. Evidently, in this case one can adjust brightness before stepping into the room. Or alternatively one can cover his watch by a hand while turning the torch on and adjusting its brightness.
2. One wishes to dim the torch in order to avoid disadaptation of eyes when making astronomic obseravtions. But still the bright default is not a problem: he can avoid looking directly on the watch while adjusting brightness, a peripheral vision is enough to do this, and hence this doesn't cause the eye's disadaptation. After the brightness is put to dim, there will be no disadaptation as well.

I can also add that there are a lot scenarios, when you need a **dim** light source, but for a **long**. In this case the dim state saves battery.

And also I would like to mention again that there are several cases when a *red light* is useful. While the middle brightness seem to be not useful.

Thefore I would prefer the following:
1. Default state in on full brightness.
2. By swipes you can change full brighness to dim - and then to red light - and then back to full brightness (i.e. in a cycle, not trying to remember should you use right or left swipe).

Riksu9000, 2022-04-14:
How about making it default max brightness on, no brightness control in app, but long pressing on the app icon in quicksettings opens it with a lower brightness?

We can keep open the possibility to change the lower brightness setting to a red light in the future.

Shatur, 2022-04-14:
> How about making it default max brightness on, no brightness control in app, but long pressing on the app icon in quicksettings opens it with a lower brightness?

Hm... Not very intuitive...

Riksu9000, 2022-04-14:
> > How about making it default max brightness on, no brightness control in app, but long pressing on the app icon in quicksettings opens it with a lower brightness?
> 
> Hm... Not very intuitive...

That's true, however once you know it, it's easy. It's the same thing with navigation. There's no indication that you can access the applications by swiping up on the watchface for instance, and that's okay. It's also not a feature everyone needs.

Shatur, 2022-04-14:
I would okay with this, but honestly, I would have brightness control as before, just enabled by default with maximum brightness. Like implemented in #1087.

medeyko, 2022-04-14:
> How about making it default max brightness on, no brightness control in app, but long pressing on the app icon in quicksettings opens it with a lower brightness?
> 
> We can keep open the possibility to change the lower brightness setting to a red light in the future.

It seems to be an interesting idea for me, but I think it shouldn't be done for the torch only, it should become a consistent secondary option for other apps as well. For example, for an alarm it could be a preset "current time + 8 hours" (as 8 hours is a default time for a sleep).

Or, if we combine, the torch may be left as it is now in 1.9.0, but long pressing on the app icon in quicksettings opens it with a full brightness on...

PR #1181 (2022-06-19): Flashlight default to max brightness
Includes tiny code tweaks.

~~Depends on #1180~~

Fixes #836
closed 2022-06-19 by JF002