InfiniTimeOrg/InfiniTime #837: Quick Replies

calhix, 2021-11-18:
I think it would be cool to be able to quickly reply to certain messages on the watch. You scroll to the message and hit a Reply button. You would then have a list of options such as yes, thanks, sounds good, later, no, etc. that would reply to the sender. Perhaps in the future you could add your own custom ones through Gadgetbridge.

Kozova1, 2021-11-28:
Android already has a feature like this for some messaging apps, perhaps it is possible to somehow reuse the suggested responses there via Gadgetbridge?

geekbozu, 2021-11-29:
I think in order for us to support this the companion app need support for it. @Avamander might know more.

JF002, 2021-12-30:
The protocol we use for the notification does not allow this kind of feature (the notification app sends the notification message/title and waits nothing in return).
Of course, we could add something similar to the phone call control (accept, ignore, reject), but that's not in our scope right now.