InfiniTimeOrg/InfiniTime #840: Phantom Touch Input Events

Batcastle, 2021-11-24:
Current Infinitime version: 1.7.0

Ever since Infinitime 1.0.0, I have been seeing phantom input events randomly occur. These most often occur for when "Raise Wrist" is enabled in "Wake Up", but I have seen them occur with that setting disabled. 

These sometimes go on for 20-30 seconds, with nothing touching the screen at all, so I find it unlikely that this is just touch events in a buffer being reacted to.

These can be annoying when you are trying to do something and the software keeps fighting you, so a fix will be greatly appreciated. Thankfully they have been reduced slightly with the new touch driver.

Thank you for all the hard work! Keep it up!

geekbozu, 2021-11-25:
I think there is/was another issue open for this, 
There is an issue with the i2c driver causing some timeouts/communication issues to occur that can cause all sorts of issues. Its on our todo list life has just been busy.

Batcastle, 2021-11-25:
Oh okay. If there is an issue then feel free to close this issue! Just wanted to put it out there in case it some weird edge case or something. Would fix it myself but I'm not good enough with C/C++ for something like that. That and I don't have a dev kit.

ajack2001my, 2022-07-04:
Just started using 1.10.0, and the problem still persist... Anybody looking into this?

JF002, 2022-07-08:
I'm not a hardware/electronic engineer, so take my words with a grain of salt, but this kind of issue if probably caused by a hardware defect... Most of the users and developer do not experience ghost touch / motion inputs so it's quite difficult to diagnose and to try to workaround with a software update.

As  @geekbozu said, we noticed that the motion sensor and/or the I²C bus seems to be easily confused by the SWD signals, but that only happens on devkit with (long) wires soldered on the SWD pads of the PCB.

Note that this only relates to "ghost inputs", when the watch is not moving and when nothing is touching the display. We are aware that the wake algorithm is a bit too sensitive and would sometimes wake the watch up even when you are not actually looking at the watch. In this case, this is a soft issue that could be fixed with a software update.

geekbozu, 2022-07-09:
More so its very very hard to test for signal integrity issues without causing them. There could be other noise sources other then SWD that we have not been able to isolate. Which can cause issues. It is just a HUGE can of worms to debug. And todo it right requires hardware, tools, knowledge, and skill that is not easy easy to get.

JF002, 2022-08-04:
I'm closing this issue as it's probably caused by an HW issue. Feel free to reopen if needed.

closed 2022-08-04 by JF002