InfiniTimeOrg/InfiniTime #841: Improved Call Notifications

Arsen6331, 2021-11-25:
Currently, call notifications are handled using the Alert Notification Service. This makes it easy to create one, but not as easy to deal with after.

For example, if someone hangs up a call before it is accepted, there needs to be a way to cancel a call notification. If there are multiple call notifications, there needs to be a way to ensure the right phone call is ended on the phone.

In my opinion, call notifications should be handled by a custom service. I think a good way to keep track of them would be to allow the companion to send an ID, and then send it back whenever a user interacts with the call notification. Also, it should be possible to send the ID again to cancel an ongoing call notification. I believe IDs should be companion-supplied rather than generated on the watch as some telephony stacks have their own call IDs that would be much easier to use, but either way would work.

Thulinma, 2021-12-05:
This is a (partial) duplicate of #367.

Riksu9000, 2023-02-21:
Closing as a duplicate of #367

closed 2023-02-21 by Riksu9000