InfiniTimeOrg/InfiniTime #842: Impossible to create a new issue using issue template

JF002, 2021-11-27:
At first, I thought it was an issue on my side, but other people have also reported this issue : when you try to create an issue using one of the templates (bug report / feature request), Github will only display a blank page when you click on the button "Submit new issue". 

I guess that's why there are very few new issues / feature requests recently.

Could somebody have a look? (pinging @nobodywasishere maybe?)

Arsen6331, 2021-11-27:
Yes, I have noticed the same thing in Firefox. I would like to add that on Chromium, I get a 400 Bad Request HTTP error code. I just created a blank issue instead.

geekbozu, 2021-11-27:
I get this as well. A quick look at the network traffic says github responds with an error 400 bad request? Kinda confusing

Avamander, 2021-11-27:
I wouldn't spend too much time diagnosing any issues related to GitHub itself right now, they're having trouble.

Arsen6331, 2021-11-27:
> I wouldn't spend too much time diagnosing any issues related to GitHub itself right now, they're having trouble.

No, this was happening before that. See when I made my "Improved call notifications" issue.

geekbozu, 2021-11-27:
> > I wouldn't spend too much time diagnosing any issues related to GitHub itself right now, they're having trouble.
> 
> No, this was happening before that. See when I made my "Improved call notifications" issue.

I agree with Arsen, This has been a bug for at least a week now :/

JF002, 2021-11-28:
I also noticed this issue 1 or 2 weeks ago, so this is probably not directly related to the outage from the previous night.

medeyko, 2021-11-28:
I experienced this problem 12 days ago (nov 16). I tried to overcome it, using different browsers and different computers, but failed. So I had to create a blank issue too.

drzacek, 2021-11-30:
Just tried opening new issue and experienced this. Can confirm

mruss77, 2021-12-01:
I had this problem on Nov. 11, and it's also mentioned in #822.

Someone on the Pinetime-dev channel said it had been happening for at least a day at that point. Looks like an issue was opened using the template on the 9th, so the problem likely began sometime between November 9-10.

I hope this bit of forensic digging is helpful!

InFerYes, 2021-12-02:
> I wouldn't spend too much time diagnosing any issues related to GitHub itself right now, they're having trouble.

I had this problem 17 days ago, even longer ago because I wrote the whole problem down and lost all my text and didn't bother until a few days later to create a new blank issue with the same "template" copied from elsewhere. So I noticed it in the weekend of the 13th/14th.

JF002, 2021-12-02:
I think I found a solution [here](https://github.com/Homebrew/discussions/discussions/2526#discussioncomment-1731168).
CLion also gave me a hint:

[image]

I committed the change in 871a673a4656c438fc12d4c0ad61d2eacc3ca340 and tested it with https://github.com/InfiniTimeOrg/InfiniTime/issues/853 : it works!

closed 2021-12-02 by JF002