InfiniTimeOrg/InfiniTime #855: Antialiased Fonts for Clock (Digital Watchface)

drzacek, 2021-12-02:
### Verification

- [X] I searched for similar feature request and found none was relevant.

### Pitch us your idea!

Pinetime would get a better, cleaner look with smooth fonts

### Description

I know antialiased fonts won't work too well on small-res fonts, but I think the huge digits of the clock on digital watchface are big enough for it to work. 

Right now we can clearly see the jagged line on the digits and for me, personally - it is repulsive. We can do better. 

So maybe someone should do it.... 

Not me though, I can't :(

kieranc, 2021-12-12:
Anti-aliasing is enabled in LVGL but the result is indeed pretty jagged, this is because the fonts are stored at 1bpp - they look a lot smoother if we increase this to 2 or 4bpp but it also takes more space. 
Since the fonts already take a large portion of our storage, this will probably have to wait until the fonts are moved to the external storage, which shouldn't be too far off now. The open_sans_light font used by pinetimestyle adds about 9kB to the firmware at 1bpp, at 2bpp it's 18kB, 4bpp 36kB...

This is how it looks at 4bpp:

[image]

It's nice, but 10% of the overall firmware size for one font. Thanks to leo341500 for testing this!

JF002, 2021-12-20:
Indeed, it does look better with anti-aliased fonts but we cannot afford the additional size for the font right now... We might be able to use better/bigger fonts when we'll use the external flash memory to store assets like pictures and fonts.

closed 2021-12-20 by JF002

TheAwesome98-Real, 2022-11-18:
the fonts are now on external storage as of infinitime 1.11.0, can this be re-opened?

JF002, 2022-11-19:
I guess it'll depend on the size of these antialiased fonts : loading fonts from external storage take a lot of time, and the fonts must be fully stored in RAM. It's worth testing though.