InfiniTimeOrg/InfiniTime #859: Building errors with cmake 3.22

Riksu9000, 2021-12-03:
### Verification

- [X] I searched for similar bug reports and found none was relevant.

### What happened?

Build fails with cmake 3.22 but works with 3.21.4

### Reproduction steps

Install cmake 3.22 -> `make pinetime-mcuboot-app`

### More details?

At first I got errors regarding missing includes in Ppg.h which doesn't occur with version 3.21.4. After fixing those, the build failed during linking. Downgrading to 3.21.4 fixed the issue.

```
[ 64%] Linking CXX executable pinetime-mcuboot-app-1.7.1.out
/usr/lib/gcc/arm-none-eabi/11.2.0/../../../../arm-none-eabi/bin/ld: cannot find -lgcc_s
/usr/lib/gcc/arm-none-eabi/11.2.0/../../../../arm-none-eabi/bin/ld: cannot find -lgcc_s
collect2: error: ld returned 1 exit status
make[3]: *** [src/CMakeFiles/pinetime-mcuboot-app.dir/build.make:1813: src/pinetime-mcuboot-app-1.7.1.out] Error 1
make[2]: *** [CMakeFiles/Makefile2:252: src/CMakeFiles/pinetime-mcuboot-app.dir/all] Error 2
make[1]: *** [CMakeFiles/Makefile2:259: src/CMakeFiles/pinetime-mcuboot-app.dir/rule] Error 2
make: *** [Makefile:189: pinetime-mcuboot-app] Error 2
```

### Version

develop

NeroBurner, 2021-12-03:
I'm currently using the following gcc version: `gcc-arm-none-eabi-9-2020-q2-update` and nRF5 sdk `nRF5_SDK_15.3.0_59ac345` and it still builds for me with CMake 3.22.0

Avamander, 2021-12-04:
You are using `/usr/lib/gcc/arm-none-eabi/11.2.0/` instead of the correct compiler.

Riksu9000, 2021-12-04:
> You are using `/usr/lib/gcc/arm-none-eabi/11.2.0/` instead of the correct compiler.

Using the recommended version makes no difference

```
/home/riku/gcc-arm/bin/../lib/gcc/arm-none-eabi/9.3.1/../../../../arm-none-eabi/bin/ld: cannot find -lgcc_s
/home/riku/gcc-arm/bin/../lib/gcc/arm-none-eabi/9.3.1/../../../../arm-none-eabi/bin/ld: cannot find -lgcc_s
collect2: error: ld returned 1 exit status
make[3]: *** [src/CMakeFiles/pinetime-mcuboot-app.dir/build.make:1813: src/pinetime-mcuboot-app-1.7.1.out] Error 1
make[2]: *** [CMakeFiles/Makefile2:252: src/CMakeFiles/pinetime-mcuboot-app.dir/all] Error 2
make[1]: *** [CMakeFiles/Makefile2:259: src/CMakeFiles/pinetime-mcuboot-app.dir/rule] Error 2
make: *** [Makefile:163: pinetime-mcuboot-app] Error 2
```

Avamander, 2021-12-04:
Nothing in CMake 3.22.0's release notes would indicate they changed its behaviour.

Very much seems specific to your setup or cmake build.

Avamander, 2021-12-04:
Not being able to find static stdlib to link against is not InfiniTime's problem until any evidence provided would indicate otherwise.

NeroBurner, 2021-12-04:
@Riksu9000 have you tried completely deleting the build directory and doing a fresh cmake-configure and build?

Riksu9000, 2021-12-04:
Deleting the build directory and reconfiguring fixed the issue and I can no longer reproduce it. I have no idea why that happened. Sorry for any confusion.

closed 2021-12-04 by Riksu9000

JF002, 2021-12-06:
@Riksu9000 I also recently had issues with building the project. I figured that adding `set(CMAKE_TRY_COMPILE_TARGET_TYPE STATIC_LIBRARY)` before the `project()` line in the main CMake file would help. I'm not sure why, but couldn't analyze that deeper.

Also, I checkout many differents branches and mostly [this one](https://github.com/InfiniTimeOrg/InfiniTime/pull/796). I probably failed one stash at some point, and couldn't build develop again (it couldn't find references to many ble functions). It turns out that the changes in the config file of lvgl.conf were still present in my working copy after the checkout to develop. Once I reset the changes, it would build again.