InfiniTimeOrg/InfiniTime #877: Bogus Heart Rate

muesli, 2021-12-13:
### Verification

- [X] I searched for similar bug reports and found none was relevant.

### What happened?

Shows bogus heart rates of >150bpm, even when sitting on a desk

### What should happen instead?

Show accurate heart rates

### Reproduction steps

I noticed that the watch often displays bogus heart rates of >150 bpm. Today I noticed it even shows that rate after receiving a notification while the watch was sitting on my desk.

### Version

v1.7.1

ajack2001my, 2021-12-13:
Similar to #532

muesli, 2021-12-13:
Agreed, looks similar. Sorry I didn't find this issue, I promise I looked for one. Closing this ticket hence.

closed 2021-12-13 by muesli