InfiniTimeOrg/InfiniTime #882: Alarm ringing causes pairing code screen to show

Itai-Nelken, 2021-12-19:
### Verification

- [X] I searched for similar bug reports and found none was relevant.

### What happened?

Every time the alarm rings (even if not connected to companion app), the pairing code screen shows.

### What should happen instead?

The pairing screen shouldn't show.

### Reproduction steps

1. Set an alarm.
  - A companion app (only tested with GadgetBridge) connected and paired or disconnected (but paired) doesn't change anything.
2. Wait for the alarm to ring.

### More details?

I haven't tested on latest develop yet.

### Version

develop 85a25302bfac215f9ec7b993f6a2c21a20ee223b with pr's #698, #720, #648, #821, #557

### Companion app

GadgetBridge 0.62.0 (from FDroid)

evergreen22, 2021-12-19:
Already fixed in develop with #874

Itai-Nelken, 2021-12-19:
Sorry, I missed that pr!

closed 2021-12-19 by Itai-Nelken