InfiniTimeOrg/InfiniTime #883: Step to conversion

Nivaddo, 2021-12-20:
### Verification

- [X] I searched for similar feature request and found none was relevant.

### Pitch us your idea!

A feature that converts steps to km/miles/calories

### Description

Whether it is by tapping on the step monitor or a seperate app that has conversions for km/miles/calories

ItzSwirlz, 2021-12-20:
The issue is, 'steps' can mean a lot; a slower person taking small steps is different from a fast person taking large steps. For someone, 1 mile could be 1000 steps but for another it could be 1500. It would have to be an 'average' conversion.

thefenriswolf, 2021-12-21:
So how do the likes of Fitbit or Apple Watch do it? Do they make you measure your steps or do they use GPS or maybe they just guess?

davidhealey, 2022-01-03:
> So how do the likes of Fitbit or Apple Watch do it? Do they make you measure your steps or do they use GPS or maybe they just guess?

They use GPS but have to be tethered to a phone - although some of the newer devices have GPS built in.

Riksu9000, 2023-02-20:
While it would definitely be nice to see the distance traveled, I'm not sure we have the capability to produce a sufficiently accurate estimation. In the end I think I'd rather not add this than add something that can potentially report wildly inaccurate distances.

JF002, 2023-02-20:
We could of course multiply the number of steps by the average step size (or even let the user configure their step size), but I'm not sure about the added value of this feature... since the step counter is triggered by many other motions than actual steps.

It might be possible to make a better estimations by processing the raw data from the accelerometer, but I'm not sure this would be in the scope of this project...

So I agree with @Riksu9000, we should not add a feature that will produce inaccurate and confusing results.

closed 2023-02-20 by JF002