InfiniTimeOrg/InfiniTime #884: Does bma421 support wakeup smart watch by interrupt?

ck-telecom, 2021-12-20:
### Verification

- [X] I searched for similar feature request and found none was relevant.

### Pitch us your idea!

support bma421 wakeup smart watch by interrupt

### Description

this should save much power when using interrupt driven

JF002, 2021-12-20:
The BMA421 does not support this feature, but the BMA425 does. IIRC, PineTimes built after July 2021 should be equipped with the 425.
Support for that feature is not implement (yet) in InfiniTime, though...

ck-telecom, 2021-12-21:
Thanks for your reply!
Hower it is a pity that BMA421 and BMA423 both are not public version prevent developer hacking.

JF002, 2021-12-21:
Yes, that clearly did help us to integrate the support for those devices...

geekbozu, 2021-12-21:
It is on the todo list. 
The 421 supports some interrupt based wakeup/low power modes just like the 425. They just need to get implemented still. As well as with the rest of the features the  425 adds to the pile.

ck-telecom, 2021-12-22:
@geekbozu I tried BMA421 data ready interrupt with my Zephyr code, found that MCU can not get been interrupted, I also poll the int register there is a data ready interrupt status. I confirmed that int1 pin was configured to output
see bellow:
https://gitee.com/fwatch/pinetime/blob/develop/drivers/sensor/bma421/bma421_trigger.c

geekbozu, 2021-12-22:
> @geekbozu I tried BMA421 data ready interrupt with my Zephyr code, found that MCU can not get been interrupted, I also poll the int register there is a data ready interrupt status. I confirmed that [...]

Great to know! This is on my to attempt to add to infinitime along with 425 support. So the notes will be more then helpful :)

ck-telecom, 2021-12-24:
@geekbozu How about your attempt :)

closed 2023-02-20 by Riksu9000