InfiniTimeOrg/InfiniTime #889: Music app not synchronising properly with device

Zaxpy, 2021-12-25:
### Verification

- [X] I searched for similar bug reports and found none was relevant.

### What happened?

Many elements and states of the media control application are not synced properly between the PineTime and Gadgetbridge

### Reproduction steps

The play time of a chosen song is frequently inaccurate and occasionally displays impossible values given the song length. 

Tapping the pause button pauses playback on the paired device however the app state does not reflect this; it continues to display as if the media is still playing, including the appearance of the pause/play button. Media is unable to be properly resumed.

### Version

v1.7.1 - cfef0131

### Companion app

Gadgetbridge 0.63.1

Avamander, 2021-12-25:
What's the music playback app you're using?

Seems like it's not sending all state changes properly.

Zaxpy, 2021-12-25:
> What's the music playback app you're using?
> 
> Seems like it's not sending all state changes properly. For example it's not sending an intent when playback has been paused.

I've only used it with Spotify, and after some quick testing Spotify is the only app I use that is effected.

digital-night, 2021-12-27:
We are experiencing the same issue with the Android Pandora App. But playback controls do work with Podcast Addict.

Avamander, 2022-01-06:
Closing as this is not InfiniTime's concern to make music players announce their status properly.

closed 2022-01-06 by Avamander