InfiniTimeOrg/InfiniTime #890: Long vibration for Notifications (for older people)

fossdd, 2021-12-25:
Hey, my family gifted my mother a PineTime yesterday at Christmas.
We've setup everything but she don't notice sometimes the vibrations of notifications incoming.
I think it would be nice to have anywhere a setting for stronger and longer vibrations for older people who don't have a very sensitive skin anymore.
Thanks! (Now my whole family has a PineTime 😎)

ITCactus, 2021-12-28:
perhaps i am old  also :) 
but i really can not feel the notification vibration when i'm walking with PineTime... 
my first thought was that the haptic/vibration motor is weak in PineTime, but several days ago i tried wasp-os, and realized, that the feedback can be really good - "Find lost device" in Gadgetbridge gives a great feel of notification. not sure if they use different pattern for vibration, or just set the different "strength"...
IMO it maybe good option to add into Settings a way to choose the "strength" for vibration.

Riksu9000, 2021-12-29:
Duplicate of #210

closed 2021-12-29 by Riksu9000