InfiniTimeOrg/InfiniTime #891: Upload failed via Gadget Bridge

Dillinger63, 2021-12-25:
I have one of the first Pine Time's with firmware 0.7.1.  I'm downloaded the latest zip file for InfiniTime and when I try and upload it via Gadget Bridge, it will get to about 7% then disconnect from the PineTime.  When I try and reconnect, it won't and the only way it will is when I reboot the PineTime.

JF002, 2021-12-25:
This is a known bug from older version of InfiniTime (< 1.6). A few bugfixes were applied in 1.6 which improved the BLE connectivity and robustness.
So I suggest you try to update to the latest version of InfiniTime. If the transfer fails, you should try to reboot your watch, your phone, try with another phone or computer, and it'll eventually succeed :-)

BTW, if you still have the old bootloader (the one that displays a green "PINETIME" logo when rebooting), you should also [update the bootloader and install the recovery firmware](https://wiki.pine64.org/wiki/Upgrade_PineTime_to_InfiniTime_1.0.0).

Dillinger63, 2021-12-25:
Yep, min still has the green Prime Time logo on boot.  I've this a go and thanks for the info, hopefully I don't brick it.

Dillinger63, 2021-12-25:
20 tries to update the bootloader and the furthest I got was 77%.  I guess I'll just buy a new Pine Time so I can update it.

Avamander, 2021-12-25:
Disabling WiFi is known to increase reliability, but there's also the option of changing the device you're attempting the upload from. 

There's also a guide on the Pine64 Wiki that describes the entire procedure, there's more to it than just updating InfiniTime.

closed 2021-12-25 by Avamander

Dillinger63, 2021-12-25:
22 tries and also disabling Gadget Bridge to not sync contacts during the firmware update did it.  I have to admit for the price of this watch, I'm pretty impressed. Now off to find some more watch faces.  Oh, I went ahead and ordered two more, one for my wife and a spare.  I really appreciate the info an help, thanks much!