InfiniTimeOrg/InfiniTime #892: Design Guidelines

PhilDevProg, 2021-12-27:
We need detailed design guidelines for InfiniTime so developers can develop more consistently. Something like what GNOME has with the Human Interface Guidelines.

Riksu9000, 2021-12-29:
There are some guidelines here https://github.com/InfiniTimeOrg/InfiniTime/blob/develop/doc/ui_guidelines.md

Generally the default theme and included fonts can be used in apps. Is there something that you think it's missing?

PhilDevProg, 2021-12-30:
@Riksu9000 In fact, I don't know... 😅 @Avamander  told me to open this issue.

Avamander, 2021-12-30:
The incompleteness of UI design guidelines was brought up in an another thread, I indeed told to solve that in a new issue.

Riksu9000, 2022-01-22:
@PhilDevProg Well, I still don't know what the issue is, so unless you have something to add, we can close this and let someone that finds an issue with the guidelines open an issue.

PhilDevProg, 2022-01-22:
> @PhilDevProg Well, I still don't know what the issue is, so unless you have something to add, we can close this and let someone that finds an issue with the guidelines open an issue.

Okay

closed 2022-01-22 by PhilDevProg