InfiniTimeOrg/InfiniTime #893: Battery Percentage Indication on Home Screen

Frowzzen, 2021-12-28:
### Verification

- [X] I searched for similar feature request and found none was relevant.

### Pitch us your idea!

Have the option to show the percentage of battery remaining on the home screen.

### Description

The battery left icon doesn't tell a person anything more than: its low. How low? Do I need a charger right this second? 

I always change phones to show the exact percentage of battery left. Seeing the data at a glance is very convenient. 

As far as I know, PineTime does not have any low battery notifications and only has the ability to show the battery icon level using missing pixels from the tiny icon. To view the percentage you need to:

1. Swipe right.
2. Click Settings.
3. Swipe up.
4. Click battery.

I feel that battery data would be more useful on the home screen next to the battery icon. Instead of being buried in the settings.

Maybe this seems like a small preference, but it really does make battery management easier when that data is right in front of you.

alastair-dm, 2021-12-29:
The circle style battery indicator is another option for a more meaningful display of battery level. See discussion [here](https://www.reddit.com/r/LineageOS/comments/bwdffk/160_new_circle_battery_icon_options_how_to_reduce/) and screenshots [here](https://i.imgur.com/JmTouva.png) with options for percentage, circle or both.

Riksu9000, 2021-12-29:
Duplicate of #582

closed 2021-12-29 by Riksu9000