InfiniTimeOrg/InfiniTime #895: Expose the HeartRate service UUID in BLE ADV packet

JF002, 2021-12-30:
### Verification

- [X] I searched for similar feature request and found none was relevant.

### Pitch us your idea!

Expose the HR service UUID in ADV so that 3rd party application detects the PineTime as a HR device

### Description

OpenTracks reported that a PineTime running InfiniTime is not detected as a HR device because the UUID of the HR service is not exposed in the ADV packet.

See https://github.com/OpenTracksApp/OpenTracks/issues/1063 and https://github.com/OpenTracksApp/OpenTracks/pull/1069.

Adding this UUID in ADV would allow companion apps to detect the PineTime as a HR device without needed to connect and discover the services running on the PineTime.

Basically, we would need to add the specific UUID here : https://github.com/InfiniTimeOrg/InfiniTime/blob/develop/src/components/ble/NimbleController.cpp#L162

[ ] What are the implications of adding multiple UUIDs to the ADV packet?

dennisguse, 2021-12-30:
Providing multiple service UUIDs in the ADV packet increases the package size and thus consumes more energy.

trman, 2021-12-30:
i don't thinks @JF002  , that the only things to do : 
even if you add the packet ,the heart rate task must be able to function even when the screen is off otherwise
(it's not the case right now)
 , this change will be  misleading since infinitime would say it's a  HR device **but** it would not show hr data only for the sporadic case when the screen is on (and for maybe only 3 fc too , since the measurement take nearly 20 sec and the screen up time is at most 30 sec)

by the way , since the waspos main dev didn't have received issue about the fact that putting the gain to 8 is bad , maybe it's time for infinitime to put the HR gain to 8 as well (in order to have better and faster measurement)i

vdmkenny, 2022-04-08:
@trman the latest infinitime firmware addresses this HR gain issue.
https://github.com/InfiniTimeOrg/InfiniTime/releases/tag/1.9.0

trman, 2022-04-08:
hi @vdmkenny 

that's true at very least  , the easiest part have been addressed (the hr gain)  and that's good !

Now it would be great if the main part would be adressed as well (add the UUI and the heart rate task function all time , even screen off)

jmaris, 2023-12-17:
No movement on this issue? Also is there anywhere we can submit bug bounties for this project?

dennisguse, 2023-12-22:
From OpenTracks perspective, the problem is resolved as it now supports to search for any BLE device as heartrate sensor (i.e., not filtering using the advertised services). 
Kind of inconvenient for the user, but works :)

florisre, 2024-07-03:
@jones139 maybe this could help resolve connectivity issues as in https://github.com/OpenSeizureDetector/PineTimeSD/issues/15?

jones139, 2024-07-03:
> @jones139 maybe this could help resolve connectivity issues as in https://github.com/OpenSeizureDetector/PineTimeSD/issues/15?

Thank you for thinking about our issue :).   I think ours is resolved - we find the HR service ok.  Now as long as the watch is not out of range for long enough for the Bluetooth on the watch to shut down, the android app reconnects to it ok.   Using a phone with Bluetooth 5 really helps the connection reliability too.

PR #2088 (2024-08-18): Advertise HR service
Builds on top of #897
The original problem was that an extra 128bit uuid was too big for the advertising packet

Works with OpenTracks (shows with the advertising filter on), NRF Connect shows both the DFU and HR service

Closes #895
closed 2024-08-18 by JF002