InfiniTimeOrg/InfiniTime #898: Export health data to Gnome Health

darkdragon-001, 2021-12-30:
### Verification

- [X] I searched for similar feature request and found none was relevant.

### Pitch us your idea!

I would like to analyze the steps and heart rate on my PinePhone. The convergent app Gnome Health could be used for this.

### Description

Gnome Health is a great and open application which is able to analyze health data in a privacy conscious way. It would be great if we can somehow get the health data from the PineTime into Gnome Health.

trman, 2021-12-30:
hi  ,
 this issue should be put first and foremost on github/gitlab.. of Gnome Health itself : 
 they need to accept infinitme device and take on it's data that are shared by bluetooth.
 (that the same tings that have been done for gadgetbridge buy an user of initime.
 
 maybe you @darkdragon-001  could check where @JF002  could retrieve the needed info , first ?

darkdragon-001, 2021-12-30:
> hi , this issue should be put first and foremost on github/gitlab.. of Gnome Health itself : they need to accept infinitme device and take on it's data that are shared by bluetooth. (that the same tings that have been done for gadgetbridge buy an user of initime.

I added issues there and linked it in the updated _references_ section.

> maybe you @darkdragon-001 could check where @JF002 could retrieve the needed info , first ?

How and where should I check what?

JF002, 2022-01-01:
Hi and thanks for your suggestion, Gnome Health looks like a nice application!
 
For now, InfiniTime exposes the "real time" value of the heart rate sensor (when enabled) and of the step count. The documentation for the BLE API is available [here](https://github.com/InfiniTimeOrg/InfiniTime/blob/develop/doc/ble.md).

We plan on adding features for fitness/health companion app (historical data, for example) but that's not available yet.

> How and where should I check what?

I'm not sure what @trman would like us to check, but feel free to ask any question you like ;-)

darkdragon-001, 2022-01-02:
> For now, InfiniTime exposes the "real time" value of the heart rate sensor (when enabled) and of the step count. The documentation for the BLE API is available [here](https://github.com/InfiniTimeOrg/InfiniTime/blob/develop/doc/ble.md).

Is this a custom API or is there some sort of standard which many different smart watches follow (would make integration in the companion apps easier this way as libraries might exist or code could be copied from other watches e.g. in Gadgetbridge)?

> We plan on adding features for fitness/health companion app (historical data, for example) but that's not available yet.

Looking forward to this!

JF002, 2022-01-02:
The HR services implements the API defined in the BLE spec. The motion service is custom as there's no standardized service for these data.

The motion (which exposes the step counter) service is documented [here](https://github.com/InfiniTimeOrg/InfiniTime/blob/develop/doc/MotionService.md).
And the HR one is [here](https://github.com/InfiniTimeOrg/InfiniTime/blob/develop/doc/ble.md#heart-rate).