InfiniTimeOrg/InfiniTime #908: Weather needs font updates

hrmckay, 2022-01-03:
With weather code being added the font jetbrains_mono_extrabold_compressed (and any other font used to display weather information) will need two more characters, the minus sign -, 0x2D, and the degree symbol °, 0xB0.

ITCactus, 2022-01-05:
is it ready? the Weather app for InfiniTime...
i saw @Avamander committed the related code about month ago, but i don't see the app on the apps list (probably for a reason)

JF002, 2022-01-05:
The weather service is merged so companion apps developers can start working on integrating it in their app, but there's no weather app yet.

@Avamander created a small demo app to test the service, but it's not included by default in InfiniTime.
I know that @kieranc is working on integrating weather info in PTS watchface.

@hrmckay is right, we'll probably need to add those characters into the font for the future weather app!

ITCactus, 2022-01-05:
> so companion apps developers can start working on integrating it in their app

just logged [feature request 2546](https://codeberg.org/Freeyourgadget/Gadgetbridge/issues/2546) about Weather sync support in Gadgetbridge

Avamander, 2022-01-05:
@ITCactus you can open the relevant code files that describe both the data format and relevant characteristics.

closed 2025-06-28 by mark9064