InfiniTimeOrg/InfiniTime #909: Option to keep display on while charging

CalcProgrammer1, 2022-01-03:
### Verification

- [X] I searched for similar feature request and found none was relevant.

### Pitch us your idea!

Keep the screen on while connected to the charger

### Description

It would be nice if there were an option to leave the screen on continuously while the PineTime is connected to external power.  Being able to dock it and have it there as a desk clock while it's charging would be nice.

Riksu9000, 2022-01-05:
This could be nice, but I would be concerned about the screen and/or the backlight wearing out faster, and I wouldn't be surprised if the components in such a cheap watch are prone to failing sooner.

Does anyone know if this would have a significant impact on the lifespan of the screen or backlight?

escoand, 2022-01-05:
Is it possible to show some content without backlight? Some smartphone doing this while loading at night.