InfiniTimeOrg/InfiniTime #913: Simple calendar view

Akselmo, 2022-01-05:
### Verification

- [X] I searched for similar feature request and found none was relevant.

### Pitch us your idea!

Calendar view that shows the current month and current day is highlighted

### Description

I just would like to see quickly from my watch what day of the month it is, or when discussing future plans it would be quicker than looking from phone. Eventually this could perhaps be used as a base for adding support for events, but right now I'd just like to see the current month and the current day highlighted. 

Also being able to scroll left and right to switch months would be nice.

Riksu9000, 2022-01-05:
I'm sure a calendar with events will come at some point. However most of the watchfaces already show the current month and day, so do you think a view of an empty calendar might still be useful in this case?

Akselmo, 2022-01-05:
I think it would, just for quickly checking what some day of some month is. "Okay next months 20th is monday, etc etc." Although if calendar with events is coming someday, making this now may be unnecessary..

medeyko, 2022-01-07:
Definitely yes, a simple calendar for current and next month, mayme with inclusion of some additional info like number of a week, full moon days, etc. is useful for a lot of people (me included, hehe).

thnikk, 2022-01-10:
I've implemented a simple calendar view with PR #923

rimugu, 2022-06-30:
Any chance we see this in a release?