InfiniTimeOrg/InfiniTime #914: dfu.py upload fails randomly

anbcodes, 2022-01-06:
### Verification

- [X] I searched for similar bug reports and found none was relevant.

### What happened?

When running dfu.py at some point through the uploading process I almost always recieve an error such as `Exception at line 159: not enough values to unpack (expected 3, got 2)` or `Exception at line 159: 'NoneType' object is not subscriptable`

### What should happen instead?

I was expecting the upload to succeed without an error

### Reproduction steps

I built InfiniTime with the enable DFU flag. I then ran

```bash
sudo ./dfu.py -z /path/to/compiled-dfu.zip -a [mac] --legacy
```

and it fails at a random time during the upload with this error

```
Exception at line 159: 'NoneType' object is not subscriptable
```

or this error

```
Exception at line 159: not enough values to unpack (expected 3, got 2)
```

### More details?

It appears that the first error is coming from this line (160) in `ble_legacy_dfu_controller.py`

```py
(proc, res, pkts) = self._wait_and_parse_notify()
```

I do not know where the second error comes from, but the error is printed on this line (167) in `dfu.py`

```py
print("Exception at line {}: {}".format(sys.exc_info()[2].tb_lineno, e))
```

### Version

develop with a slight change of a string ('+' -> '^' in timer.cpp)

anbcodes, 2022-01-06:
After asking on the matrix PineTime channel, KieranC said that some recent changes are causing problems with dfu because they are running out of memory, so that appears to be the source of the issue

JF002, 2022-03-13:
I used the script `dfu.py` to test and debug DFU when I was implementing it and when no companion apps were available. I took this script from [WaspOS](https://github.com/daniel-thompson/ota-dfu-python/blob/3d6fd30d33c2b20bc86ff6b9269fddf4a1d4c7c6/dfu.py) and slightly modified a few delays.

Since then, we have many companion app that implement the DFU protocol and that script is not maintained.

Unless someone is willing to maintain it, I'll probably remove it to avoid confusion.