InfiniTimeOrg/InfiniTime #917: {Question] Milestone 1.8.0 is complete

ajack2001my, 2022-01-07:
Is there is reason why 1.8.0 hasn't been released yet? Asking as all milestones have been met.

I am eager to update my PineTime as current build has some annoying bugs (see my bug reports) and I hope the 1.8.0 fixes it.

Riksu9000, 2022-01-07:
It will be released when we feel it's ready. Milestones aren't necessarily representative of progress towards a release as things can always change. Please be patient :)

closed 2022-01-07 by Riksu9000

JF002, 2022-01-07:
I had originally planned to release 1.8 yesterday but had to deal with private matters instead sooo... I've just released it a few minutes ago  : https://github.com/InfiniTimeOrg/InfiniTime/releases/tag/1.8.0. Enjoy!