InfiniTimeOrg/InfiniTime #921: Simultaneous notification and Chime causes long, intense vibration, then watch crashes

jenb82, 2022-01-08:
# Specs
Infinitime 1.8.0
Android 9 (LineageOS 16)
Gadgetbridge 0.64

## Settings on watch:
- Chimes every 30 minutes
- Vibration on
- I JUST updated to 1.8.0

## Settings on phone:
- Connected to Gadgetbridge (I have no set up secure connection yet)
- I have an app that sends me a notification every hour, on the hour

# **What happens:**
At :00 minutes, I simultaneously receive a Chime and a notification from my phone. 
- The watch screen does not turn on
- The watch starts vibrating VERY aggressively
- The watch continues to vibrate for ~5 seconds
- The watch reboots

**This does not happen every time.** I installed 1.8.0 seven hours ago, and it has happened twice. 

I'll report back if I notice other details. 

Otherwise, 1.8.0 is AMAZING. Thanks for your work!

SteveAmor, 2022-01-08:
@jenb82 thanks for the information. The code doesn't check for a notification collision as I assumed the code architecture would take care of that. One situation I did cover off was chimes not switching the display to the clock face and vibrating if the alarm is alarming (I.e. alarm is a higher priority than chimes).  I'm hoping someone with a better understanding of the architecture might have a good idea of this race condition and a potential fix to try.
Would you be able to switch chimes off and set an alarm on the hour to see what happens when your hourly app causes a notification to be sent at the same time the alarm goes off?
Is your hourly notification app available for others to test with?
It would be good to resolve this bug as it might recurr in another situation too. There's always a chance that a user will get one of these edge case collisions for other asynchronous notification that the companion app sends exactly on the hour/half hour.

jenb82, 2022-01-08:
@SteveAmor thanks for the response and explanation!    I'll test the alarm setup you described this weekend and let you know what happens.  

The notification app is "Mindful Notifier" on F-Droid. On the Schedule menu, I have "Periodic" "1:00 hour" reminders, with Quiet Time beginning and ending at a X:00 time.  On the home screen for the app, it is "enabled" with "vibrate" on.  https://f-droid.org/en/packages/com.kmac5dev.mindfulnotifier/

And to be clear, this is not a high priority bug. I can change my hourly notification to X:01 without any complaints.  It's also possible I'm misinterpreting what's causing the watch to crash, since the problem has only happened twice.

KoalaV2, 2022-01-08:
+1, Same problem. If I get a notification and chimes reminder at the same time i get a long vibration and then a crash. Running latest git commit.

jenb82, 2022-01-10:
@SteveAmor Here's an update on my testing

# What I Did:
1. Turned Chimes off
2. Kept my hourly phone notification
3. Repeatedly set an alarm for X:00, to coincide with the notification

# What Happened:
At X:00
1. The watch vibrates once, and displays my phone notification
2. Once the notification disappears, my watch moves to the Alarm App.  The Alarm is triggered, with a red Stop button, but in 6/7 cases, **the alarm did not trigger an ongoing vibration**.  In the seventh case, the alarm started vibrating after a few seconds. 

**So far, this has not triggered the problem described in this thread.** There is no long vibration, and the watch does not crash.

Riksu9000, 2022-01-10:
I wonder if this is due to low memory or memory corruption, because it is inconsistent, and I had a similar issue where the watch didn't wake up and kept vibrating, but it was triggered by something else.

NeroBurner, 2022-03-09:
maaaaaaybe this issue is related to https://github.com/InfiniTimeOrg/InfiniTime/pull/1026

if someone could test with the latest develop branch as well, that would be awesome

JF002, 2022-03-13:
I spent a couple of hours trying to reproduce this issue, with no luck...
First, I tried to generate the events internally so that a chime and a notification would be generated at the same time.
Then I changed the timing so that a chime would be generated every 10s, leaving enough time for the watch to go to sleep between each tried. I would then send a notification using a companion app when the chime was generated.

I did all this on 1.8.0 (without #1026) and it's never crashed...

Does anyone still experience this issue? Can you provide any additional info that could help debugging?

NeroBurner, 2022-03-13:
have you opened the chimes settings screen before doing the tests? Maybe those mangle some memory which causes an issue some time later

jenb82, 2022-03-14:
I'll spend some time trying to reproduce my original problem and report back.  I turned off chimes after posting this bug, so I don't have more data to offer.

JF002, 2022-03-14:
> have you opened the chimes settings screen before doing the tests? Maybe those mangle some memory which causes an issue some time later

I did this at least once, when I checked if Chimes were enabled.

This issue looks more like a race condition than a memory corruption, IMO, but I can be wrong :). 
I noticed that the motorcontroller is used in SystemTask and also in DisplayApp. In this case, both task could access to the motorcontroller at the same time, and it's not reentrant or thread-safe...

jenb82, 2022-03-15:
No luck reproducing the problem today, but I'll keep trying.  

In case it helps, more information:

- I had the issue right after upgrading 1.7.1 > 1.8.  I haven't touched the firmware since. 
- I was not using Secure Pairing when I had the issue.  Now I am. Perhaps the encryption adds enough delay to the process to offset the two events? (i.e. the companion app notification is no longer EXACTLY simultaneous) If I can find a way to test this without wiping my Gadgetbridge history, I will.    Not sure if others in this thread were using Secure Pairing.

mark9064, 2025-10-20:
The motor controller is only used in DisplayApp now (I haven't looked for when this changed but definitely a while ago), so this should be solved

closed 2025-10-20 by mark9064