InfiniTimeOrg/InfiniTime #922: Better watch face creation and adding

TotalChaos05, 2022-01-08:
### Verification

- [X] I searched for similar feature request and found none was relevant.

### Pitch us your idea!

I think that a simple gui app to make watchfaces and a way to add them without rebuilding firmware would help

### Description

I thought it would give people more of an incentive to make watch faces if it was easier to make and install them. I know that this watch is geared more towards developers, but I feel like this would attract more casual users.

Danimations, 2022-03-31:
As a designer, I really like this idea. A potential next step could be simply allowing for the transmission via Bluetooth of a 240x240 pixel BMP or similar to use as a background behind an existing watch face.

Avamander, 2022-03-31:
BMP as a format is out of the question because it's very incompatible with our hardware limitations.

Avamander, 2022-03-31:
Duplicate of https://github.com/InfiniTimeOrg/InfiniTime/issues/755

closed 2022-03-31 by Avamander