InfiniTimeOrg/InfiniTime #925: Cannot DFU via GadgetBridge

danielbarry, 2022-01-09:
### Verification

- [X] I searched for similar bug reports and found none was relevant.

### What happened?

Cannot upgrade from 1.8.8 with the DFU, GadgetBridhe complains 'No Element found'.

### What should happen instead?

PineTime updated via DFU.

### Reproduction steps

Pair device via GadgetBridge, download DFU from releases, open with GadgetBridge and attempt to upload. I have also tried restarting the device several times just before upload. Tried with multiple devices too.

### More details?

* GadgetBridge 0.62.0 (tried with latest version too)
* InfiniTime 1.5.8 (appears to have reverted to older version)
* Tried to upload 1.5, 1.6, 1.7, 1.8 - all failed.

### Version

v1.5.8

### Companion app

GadgetBridge 0.62.0

Strit, 2022-01-09:
I had this issue for months. My workaround is to use nRF Connect to update the firmware. That seems to work just fine.

JF002, 2022-01-09:
InfiniTime 1.5.8 is not part of the official release; Did you mean 1.5.0?
Also, make sure you open the **dfu (.zip)** file and not the **binary (.bin)** file. Gadgetbridge expects that zip file that contains the firmware + some meta data. The .bin file can be opened by Amazfish, for example.
Here's the link to the correct file for Gadgetbridge to update to 1.8.0 : https://github.com/InfiniTimeOrg/InfiniTime/releases/download/1.8.0/pinetime-mcuboot-app-dfu-1.8.0.zip.

danielbarry, 2022-01-10:
> I had this issue for months. My workaround is to use nRF Connect to update the firmware. That seems to work just fine.

I think I can give this a go.

> InfiniTime 1.5.8 is not part of the official release; Did you mean 1.5.0?

It might just be the font, not sure if zero or eight on the end. The ShortRef is 6652ec6f

> Also, make sure you open the **dfu (.zip)** file and not the **binary (.bin)** file. Gadgetbridge expects that zip file that contains the firmware + some meta data. The .bin file can be opened by Amazfish, for example.

As I said I've been successfully flashing the device for quite a while now. I even went back to a firmware I had successfully flashed in the past.

danielbarry, 2022-01-10:
@Strit Appears to be working with the nRF Connect app at the grand speed of 1.2kB/s.

GadgetBridge cannot even connect to it any more, it's just acting weird.

ajack2001my, 2022-01-10:
I believe the 1.8.0 firmware requires Gadgetbridge users to use GB 0.64.0 to updated the PineTimes... I did that, no problem.

dadall, 2022-01-10:
@ajack2001my The last line of the release notes : 

> Note to Gadgetbridge users : Please update to the latest version of Gadgetbridge (0.64.0 from F-Droid) before trying to update InfiniTime. We recently found an issue in Gadgetbridge 0.63.1 that confused the OTA in InfiniTime. Version 0.64.0 will fix the error message at the beginning of the OTA.

:wink:

danielbarry, 2022-01-10:
> @ajack2001my The last line of the release notes :
>
>
> wink

Maybe the problem is that I used version 0.62.0, which then confused the PineTime OTA? I tried 0.64.0 afterwards, but it was not working.

In any case, the nRF Connect app worked as @Strit recommended. I'll close this issue and hopefully it will help others which find themselves in a similar spot.

closed 2022-01-10 by danielbarry