InfiniTimeOrg/InfiniTime #927: hardware button to put watch to sleep causes instant wake under specific circumstances

Zaxpy, 2022-01-10:
### Verification

- [X] I searched for similar bug reports and found none was relevant.

### What happened?

Pressing the hardware button while holding the watch vertically and with shake to wake enabled with a low sensitivity (<50%) causes the watch to immediately wake up.

### Reproduction steps

- hold the watch vertically (with the hardware button facing up or down)
- press the hardware button on the watch face as if you were putting it to sleep

### More details?

- setting the sensitivity to over 50% prevents this issue
- lower sensitivities can cause the issue to occur at less extreme angles (closer to being held horizontally)

### Version

v1.8.0

### Companion app

Gadgetbridge 0.64.0

fossdd, 2022-03-24:
!'
2!! '2

'# @
. @ sq

fossdd, 2022-03-24:
! @.!'!
!

fossdd, 2022-03-24:
' h.

fossdd, 2022-03-24:
?ZHh

fossdd, 2022-03-24:
Q!?qwQqZw. Z 9.bvVWZqbO
0@&!

OskarSchamardin, 2022-10-21:
Confirmed bug still present in v1.11.0, this can be visualized by holding watch with button facing the floor/ceiling and opening the 'shake calibration' settings menu. The red bar jumps to 100%.

isaacc27, 2023-03-16:
I believe I've found the issue. In the ShouldShakeWake() function when doing the calculation for the speed, the previous value of x is never taken into account. Therefore, when for example the watch is held with the button facing towards the ceiling, it miscalculates that the speed is a constant ~185 when it should be 0. This is exacerbated by the fact when the button is pressed to put the device to sleep, UpdateMotion() seems to be called in quick succession (I still don't know why, can anyone provide insight?), thus messing up the speed calculation since the timing is severely shortened and speed greatly increases.

As a side note, I've also noticed that this error doesn't trigger if the button is pressed very slowly while the button is facing towards the ceiling.

PR #1691 (2023-06-17): ShakeWake: Fixed instant wake after sleep issue in certain positions
Added lastX var to track the previous x acceleration for correct calculation of speed. 
Closes #927.
closed 2023-06-17 by JF002