InfiniTimeOrg/InfiniTime #928: Tracking steps between Gadgetbridge and PineTime seems inaccurate...

ajack2001my, 2022-01-11:
### Verification

- [X] I searched for similar bug reports and found none was relevant.

### What happened?

I walked and GB reported I took 536millions steps by mid day

### What should happen instead?

Typically around 5k steps by mid day

### Reproduction steps

Don't know...

### More details?

[image]
[image]

### Version

1.8.0

### Companion app

GB 0.64.0

ITCactus, 2022-01-11:
i have never experienced such an issue, even during implementing steps sync support for Gadgetbridge.
could it be some issues with Gadgetbridge in "Settings -> About you"?
also, did you try to remove device from Gadgetbridge, and do a clean pair like a new device...

anyway, it's unlikely the InfiniTime issue, but anyway can you look in Gadgetbridge logs e.g. in Android Studio?
step sync should look like this:
`D/nodomain.freeyourgadget.gadgetbridge.service.btle.BtLEQueue: characteristic changed: 00030001-78fc-48fe-8e23-433b3a1942d0 value: 0x08 0x00 0x00 0x00
D/nodomain.freeyourgadget.gadgetbridge.service.devices.pinetime.PineTimeJFSupport: onCharacteristicChanged: MotionService:Steps=8
D/nodomain.freeyourgadget.gadgetbridge.service.devices.pinetime.PineTimeJFSupport: adding 8 steps
`
do you see correct values, or those are already in thousands?

@JF002, @Avamander, just wondering if it could be a side effect of  the feature mentioned as ["secure pairing"](https://github.com/InfiniTimeOrg/InfiniTime/releases#:~:text=For%20now%2C%20the,secure%20pairing%20mode!) ?

Avamander, 2022-01-11:
@ITCactus unlikely to be the cause.

Must've been a communication error somewhere, given the fact that it's nonreproducible I don't think we can do anything here.

closed 2022-01-11 by Avamander

ITCactus, 2022-01-12:
well, for the first time i noticed some issue with steps counter - today PineTime/InfiniTime counted only about 400 steps instead of couple of thousands. and no steps info on charts in Gadgetbridge at all...
unfortunately, that version of Gadgetbridge is not debugable, and all i can see is logs (after enabling those in GB settings). and in logs i see that sync steps from InfiniTime  to Gadgetbridge is ok, but those are not added for some reason.

aereaux, 2022-01-24:
I've been experiencing some issues like this, where the value in gadgetbridge will be significantly more than the value on the pinetime.  It happens somewhat often, although I haven't found a way to definitively reproduce it.  Is there anything I can do to help debug this?