InfiniTimeOrg/InfiniTime #929: No response from hardware button

Mentate, 2022-01-11:
After updating to 1.8.0 the hardware button fails to respond at all. It no longer has any effect on the watch, including the long press to shut down feature.

Behavior:
No response when hardware button is pressed. 

Expected Behavior:
For the watch to respond when the button is pressed

After a rollback to the previous firmware, the hardware button began working again.

Avamander, 2022-01-11:
If you repeat the update, does it reoccur?

Mentate, 2022-01-11:
I installed it on my dev watch first, and when the button failed I assumed I had messed up the hardware so I installed it on my main watch and had the same results.

Mentate, 2022-01-11:
I should mention that I compiled it myself.

JF002, 2022-01-19:
@Mentate  Could you try with the build [from the release note](https://github.com/InfiniTimeOrg/InfiniTime/releases/tag/1.8.0) to eliminate the possibility your build is faulty?

honyczek, 2022-02-03:
Hi, I've updated my watch today from official 1.7.1 to official 1.8.0 and after successful download of firmware watch went to reboot, but never started up. Hardware button doesn't work as hard reset nor power on. After trying to charge watch it is still off. I have Gatgetbridge 0.65.0 and sealed PineTime...

Riksu9000, 2022-02-21:
@Mentate Did you try the official release build? Do you still have this issue?

Mentate, 2022-03-08:
Sorry for the bad delay in response, using the official repo corrected the issue, must have been something wrong with my build change

closed 2022-03-08 by Mentate