InfiniTimeOrg/InfiniTime #931: Status bar shows wrong time format

Mentate, 2022-01-12:
### Verification

- [X] I searched for similar bug reports and found none was relevant.

### What happened?

The time in the status bar above the app list page is in 24 hour format

### What should happen instead?

It should be in the same format as the settings specify

### Reproduction steps

Go to settings and set the time to 12 hour format. Go to the apps list and in the top status bar the time is still in 24 hour format. 

### Version

1.7.1

### Companion app

GadgetBridge

moriel5, 2022-01-13:
While I don't believe that this is a bug, but simply an unimplemented (but planned), yeah, I would prefer it to show in a 12 hour format.

Currently, I believe that the toggle only changes the format on the watchface/main screen.

moriel5, 2022-01-13:
@Itai-Nelken Thanks, I guess in that case that is either a bug or an oversight?

closed 2022-02-13 by JF002