InfiniTimeOrg/InfiniTime #934: notification are sometimes out of order

42Tobias42, 2022-01-14:
### Verification

- [X] I searched for similar bug reports and found none was relevant.

### What happened?

sometimes the order of notification is wrong. When i swipe down, it shows the first notification. But swiping back to the watch face brings up the 5th notification

### What should happen instead?

it should show the notifications in order and go back to the watch face.

### Reproduction steps

No idea - played around with test notifications, but couldn't reproduce it.

### More details?

So the normal order should be:
watchface, 1, 2, 3, 4, 5

The order i get is:
watchface, 5, 1, 2, 3, 4 (for example - sometimes it shows 5,4, 1,...)

Swiping down from the watchface brings me to 1, but swiping back up shows notification 5.

After 5 notifications the order is normal again.

### Version

infinitime 1.8.0, bootloader 1.0.0

### Companion app

Gadgetbridge 0.64.0

Itai-Nelken, 2022-01-15:
I also noticed it but it only happened 2 or 3 times and I thought it probably was an issue with the custom build I was using.

calhix, 2022-01-17:
Slightly unrelated, I sometimes get 2 notifications with the exact same text when it should only be one notification, maybe 50% of the time.

SteveAmor, 2022-01-19:
I get two notifications with the same text for SMS. One with the phone number and one with the Contact name. But that is GadgetBridge, not Infinitime.

JF002, 2022-01-19:
Do you have any hint on how we can reproduce this (duplicate and out of order notifications)?

42Tobias42, 2022-01-19:
out of order notification:
not really - I tried:
- receiving a notification while I'm in the notification menu
- receiving long notifications

what I haven't tried:
- receiving a lot of notifications in a row (or at the same time)

And it normally happens when i get messages in Signal

calhix, 2022-01-23:
Also on Signal, I haven't really noticed if notifications have been out of order but it's just a bit random when it comes to duplicates. Actually I wonder if it was incoming messages from someone that doesn't have Signal and I have the app set as the unified SMS messenger. I'm not sure but I will update when it happens again/try to get someone not on Signal to send me an SMS.

JF002, 2022-03-13:
Unfortunately, I couldn't reproduce this issue. I'll close this issue for now.

Feel free to re-open if this issue still occurs and/or if you can provide more info to help debugging :)

closed 2022-03-13 by JF002