InfiniTimeOrg/InfiniTime #935: Stopping Alarm Vibration

bokf, 2022-01-14:
### Verification

- [X] I searched for similar feature request and found none was relevant.

### Pitch us your idea!

The alarm vibration should stop automatically after some seconds

### Description

The alarm function is great, but the signal is "never" ending and very disturbing after a short time. If I need a loud alarm I can take a mobile phone or a traditional alarm clock. One benefit of using the alarm of the Pinetime is to have a nearly silent alarm clock on the wrist to not wake up other people in the same room.

- I would prefer that the alarm vibration is stopping after 10 sec automatically without any user interaction. People that are not awake after 10 sec vibration do also not wake up after 10 min vibration on the wrist? In case 10 sec are too short for some people, maybe a reminder after 10 min would be an idea, if the alarm was not switched of manually by the user?

- I would prefer that I can switch of the alarm also by pressing the mechanical button at the side. Searching below the blanket for the right little button on the screen takes me too long in the morning and all other people in the room are also awake.

Thanks for considering.

dyamon, 2022-01-16:
A few ideas:

- the first point seems a bit weird to me, making the Alarm app something that "wakes some people, but not if you are a heavy sleeper". Maybe making it a 1m alarm that repeats every 10m is a good in-between?

- on the second point: the idea is usually to make stopping the alarm a semi-challenging task (e.g. on a phone you need to swipe a particular area). I would maybe use the side button for snooze?

bokf, 2022-01-16:
I think we should have some more feedback from people that really use Infinitime to wake them up. After which time period they woke up. For me already 10sec is much too long. I wake up after ca. 2 sec i.e. 2 vibrations. And I don't think that heavy sleepers wake up after 5 min of vibrating, but let's hear them.

bokf, 2022-01-16:
Or something like 10 sec vibration and than sleep for 5 min and repeat this 10 times. There are many possibilities. But the only solution that is not allowed is an alarm that sounds forever.

Itai-Nelken, 2022-01-16:
personally I like that it never stops. it's the only watch alarm I tried that can wake me up.
of course it isn't practical to have it ringing forever, but I think 10 seconds is a bit too short.
maybe 30 seconds followed by X minutes and repeat Y times?
pressing the physical button could stop the current 30 seconds and wait for the next (snooze).

Riksu9000, 2022-01-17:
There doesn't seem to be an issue with the alarm being neverending, as long as it can be stopped by pressing the button, which I agree it should do.

bokf, 2022-01-17:
I don't think that two personal opinions are an objective survey regarding never ending alarms. From my point of view it is a no-go. The running vibration for hours is a useless drain of battery. Imagine you have a daily alarm at 5 o'clock in the morning and leave your watch unattended for a day (I didn't wear watches for many many years and I don't need a watch every day) the alarm can wake up several apartments, if it is laying on a night table with a little bit of resonance properties. I wake up frequently by those alarms from other people for example in hotels it is horrible. I don't see the advantage of an alarm running forever.

But I can understand Itai-Nelken, if he wishes 30 sec ringing. Last night I tried to abuse the timer as an alarm and in fact it did not wake me up during deep sleep phase. Unfortunately maximum setting is 59:59 min, so I couldn't use it as alarm replacement for testing purpose.

So the actual proposal is:
Ring 30 sec, stop for 5 min, ring 30 sec and repeat this 6 times. This means the alarm tries to wake you up during 30min. An alarm ringing longer is useless, because most probably you missed your appointment anyway ;-)

Addionally:
"pressing the physical button could stop the current 30 seconds and wait for the next (snooze)."

bokf, 2022-01-18:
As a mechanical engineer I can also tell, that we try to avoid vibrations wherever possible. Vibrations have a great destructive potential. Also chips and soldered parts are not immune against it. So reducing the time where the PineTime vibrates will increase its lifetime.

InFerYes, 2022-01-21:
> I think we should have some more feedback from people that really use Infinitime to wake them up. After which time period they woke up. For me already 10sec is much too long. I wake up from the first vibration. And I don't think that heavy sleepers wake up after 5 min of vibrating, but let's hear them.

I use it to wake up, because my regular "sounding alarm" drove people nuts. I can't tell how long it was vibrating before it wakes me up and I turn it off because at that point I'm still to drowsy to realize what's going on, but it needs to keep going or it won't wake me up. 10 seconds as suggested is too short probably.

ThePinkUnicorn6, 2022-01-30:
Would it be possible to just have a settings page with a few options to cater for different needs? Such as a 10 second option, a 30 second option, 1 minute option, and an option so it wont stop until interacted with.

warnerbryce, 2022-02-01:
I use the PineTime for wake me up the morning. 
I don't wear it at night, the watch is on my desk. 
When it rumbles, it wakes me up immediately because my desk amplify the noise.

But like a real alarm, it should vibrate during 1 minute then stop during X minutes, then vibrate again. Without stopping since the button is tapped.

On another topic (off topic) the TimerApp should vibrate during 30 seconds if nobody touch the watch. Just one vibe is so subtle an i often miss the end of the timer.