InfiniTimeOrg/InfiniTime #936: Timer Countdown Signal is too weak

bokf, 2022-01-14:
### Verification

- [X] I searched for similar feature request and found none was relevant.

### Pitch us your idea!

The end of the timer countdown should be more remarkable

### Description

Opposite to the never ending vibration of the alarm clock, the vibration of the countdown timer is much to short and weak. It just makes a short "Bzz" and thats all. If I am walking or moving I often miss this signal. If there would be a vibration countdown the last couple of seconds with short vibrations and a longer vibration at the end (like "bzz bzz bzz bzz bzzzzzzzzz") it would be much better to notice the countdown end. I did use this timer only once for baking my bread because it was much to weak and I missed it ...

Itai-Nelken, 2022-01-15:
#557 addresses this issue.

Riksu9000, 2022-01-15:
Duplicate of #391

closed 2022-01-15 by Riksu9000