InfiniTimeOrg/InfiniTime #942: no communication between watch and app after bluetooth connection

craigsickmen, 2022-01-17:
### Verification

- [X] I searched for similar bug reports and found none was relevant.

### What happened?

I updated to 1.8.0 using Gadgetbridge 0.64.0, and found that despite bluetooth connection, no information seems to be sending between the two devices.

### What should happen instead?

Information like notifications and current tract should be communicated to pinetime, and data from infinitime should be sent to gadgetbridge.

### Reproduction steps

Updated to 1.8.0 from 1.7.0 using Gadgetbridge 0.64.0. On reverting to 1.7.0, functionality returns.

### More details?

Using the debug menu in gadgetbridge, I confirmed that infinitime was not receiving any notifications. I also could not control media on android using the music app in infinitime.

### Version

v1.8.0

### Companion app

Gadgetbridge 0.64.0

SteveAmor, 2022-01-17:
Samsung A40 with 0.64.0 and 1.8.0 seems to work OK.

ITCactus, 2022-01-17:
sometimes the good option is to remove pairing in Android Bluetooth settings, and do the pairing again, especially in case of upgrade to 'InfiniTime' 1.8, where secure pairing (required PIN) was introduced.
also (based on experience with 'wasp-os'), sometimes there is no other option then remove device in 'Gadgetbridge', and add it again.

craigsickmen, 2022-01-19:
@ITCactus Thanks for the feedback, gave both of these options a try with no success. I believe it might be an issue with my device.

Avamander, 2022-01-19:
This seems isolated to your setup. Do try just removing all remembered/paired devices from your phone, restart your phone and see if that helps.

JF002, 2022-01-26:
@craigsickmen Any update about this? Did you try to connect to the watch without pairing/bonding? Gadgetbridge opens a prompt asking if you want to pair when it connects, did you try to select "don't pair"? This would allow us to check if the issue comes from the secure pairing/bonding.

craigsickmen, 2022-01-31:
@JF002 I attempted to connect without pairing/bonding, no change in behavior. Also, now even when I revert back to 1.7.0, functionality does not return.

Baggypants, 2022-02-03:
Hi there, I have a similar issue. I have a new sealed pinetime. I managed to get it updated from 1.6.0 to 1.8.0 after a reboot with siglo, but unless I keep rebooting I can't get it to stay connected to either Gadgetbridge or siglo for more than a few minutes until I reboot the watch again.

Baggypants, 2022-02-05:
I loaded on 1.7.1 and that works.

Oblomov, 2022-02-10:
Additional information: after reinstalling the firmware and rebooting, the device shows the BT icon, but GB is stuck in “connecting”, although it does show information about the hardware and firmware versions now.

JF002, 2022-02-12:
> (1) the device remains with the PIN showing (until I press the button) 

This is expected in current version: the PIN screen is not closed automatically when the connection is established. 

> OK, one last piece of information: after validating the FW and rebooting one more time, the issue seems to have been solved. (OTOH, I don't like the fact that I had to validate it BEFORE having it work correctly ...)

The validation process should not have any impact on the BLE connection (it just writes a byte in flash memory so the bootloader knows the FW is validated). So I guess it works because you rebooted your watch.

To be honest, I don't know why it wasn't working before and why it suddenly worked... BLE sometimes acts strangely, and the result differs from one device to another, which makes the integration very difficult.
One think you can check : did you ensure that no other BLE device (phone, computer,...) was trying to connect to your PineTime at the same time than your phone? That would explain this "lottery" game each time your reboot your pinetime.

Oblomov, 2022-02-12:
> This is expected in current version: the PIN screen is not closed automatically when the connection is established.

Oh, OK. I thought this was an issue similar to the call notification remaining even after the phone has been answered.

> To be honest, I don't know why it wasn't working before and why it suddenly worked... BLE sometimes acts strangely, and the result differs from one device to another, which makes the integration very difficult.

Shouldn't the devices (either side) be able to detect that the connection isn't actually working, though? The weird situation in this case is that the failure isn't a failure to connect or pair, like it used to be with older firmware, but an allegedly successful pairing where no further communication actually manages to successfully complete.

> One think you can check : did you ensure that no other BLE device (phone, computer,...) was trying to connect to your PineTime at the same time than your phone? That would explain this "lottery" game each time your reboot your pinetime.

There were no other BLE devices in range other than my phone and the watch, and again the issue wasn't so much an inability to pair, but rather a somehow non-functional pairing.

JF002, 2022-02-13:
> Oh, OK. I thought this was an issue similar to the call notification remaining even after the phone has been answered.

It could probably be improved to be less confusing :)

> Shouldn't the devices (either side) be able to detect that the connection isn't actually working, though? 

Theoretically, yes. In practice, the BLE stack is a really complex piece of code and the results vary from one 'companion' (phone, computer) device to another, probably because their own BLE stacks are different and expose different behaviors and eventual bugs.

craigsickmen, 2022-02-14:
this morning I decided to try flashing 1.8.0 again, and for some reason, it just worked. Same version of gadgetbridge,  same DFU and image files. @JF002 would you like me to close the issue? It seems to have garnered activity from others.

JF002, 2022-02-19:
@craigsickmen You've probably experienced a glitch from the update and the addition of the secure pairing mode... I'm happy it's now working fine for you! I'll close this issue. Feel free to reopen if needed :)

closed 2022-02-19 by JF002