InfiniTimeOrg/InfiniTime #943: Shake return to main screen

rostislavYurievich, 2022-01-17:
### Verification

- [X] I searched for similar feature request and found none was relevant.

### Pitch us your idea!

Use shake gesture to return to main screen

### Description

It resolves issue, when you want to see the time, but there is a some menu open. To close this menu you need to use your second hand.

ItzSwirlz, 2022-02-03:
Shaking would be an issue because if I'm playing 2048 or Pong or whatever, and I move my arm it can be problematic.

What I am trying to look into, since my PineTime button broke/fell off, as I wait to 3D print it, is to have a 7 second touch screen to return to home.

Riksu9000, 2023-02-23:
#1592 tracks the issue where you can't see the time in apps.

closed 2023-02-23 by Riksu9000